chipsalliance/riscv-dv #387: Support directed assembly tests in testlist.yaml

taoliug, 2019-12-24:
Current testlist format:
```yaml
- test: riscv_jump_stress_test
  description: >
    Stress back-to-back jump instruction test
  iterations: 2
  gen_test: riscv_instr_base_test
  gen_opts: >
    +instr_cnt=5000
    +num_of_sub_program=5
    +directed_instr_1=riscv_jal_instr,20
  rtl_test: core_base_test
```
Suggest to add a new field asm_test to allow user specify directed assembly test in the testlist. This will enable user to run random instruction test and directed assembly test in one regression.

Sample test entries
```yaml
- test: riscv_single_assembly_test
  description: >
     single assembly test entry
  iterations: 1
  asm_test: sample_assembly.S

- test: riscv_assembly_regression_test
  description: >
    Run all assembly tests under the given directory
  iterations: 1
  asm_test: assembly_test_directory
  gcc_opts:
     # Some custom gcc options
```

taoliug, 2019-12-24:
@danghai FYI.

udinator, 2019-12-27:
Addressed in #391

closed 2020-01-11 by taoliug