chipsalliance/riscv-dv #393: Support for RISCV Vector extension

Sustrak, 2019-12-28:
Altough the vector extension for RISCV has not already get to the version 1.0 is there any plan to incorporate vector instructions to the code generator ? I've seen that the structure for it is more or less already defined. What would it be necessary to implement it ?

Thanks

taoliug, 2019-12-29:
Adding vector extension is included in the 2020 H1 development plan. Some preparation work is done:
- Refactored the instruction class to make  adding new ISA extension (B/V) more modular.
- Refactored the trace CSV format to make it scalable to support large number of new architectural states (vreg) introduced by V extension.

I'd say the framework is ready, just need to pick a stable spec/toolchain version (0.71 for now) to start the implementation.

taoliug, 2019-12-30:
Closing this issue for now, we will file issues to track the implementation progress once we start working on this.

closed 2019-12-30 by taoliug