chipsalliance/riscv-dv #397: GCC issue?

weedmank, 2019-12-29:
I tried running: python3 run.py --test riscv_arithmetic_basic_test --simulator questa
and got the following.

[engr@localhost riscv-dv]$ python3 run.py --test riscv_arithmetic_basic_test --simulator questa
Sun, 29 Dec 2019 13:42:22 INFO     Processing regression test list : /home/engr/riscv-dv/yaml/testlist.yaml, test: riscv_arithmetic_basic_test
Sun, 29 Dec 2019 13:42:22 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:2
Sun, 29 Dec 2019 13:42:22 INFO     Processing simulator setup file : /home/engr/riscv-dv/yaml/simulator.yaml
Sun, 29 Dec 2019 13:42:22 INFO     Found matching simulator: questa
Sun, 29 Dec 2019 13:42:22 INFO     Building RISC-V instruction generator
Sun, 29 Dec 2019 13:42:22 INFO     Running RISC-V instruction generator
Sun, 29 Dec 2019 13:42:22 INFO     Generating 2 riscv_arithmetic_basic_test
Sun, 29 Dec 2019 13:42:22 INFO     Running riscv_arithmetic_basic_test with 1 batches
Sun, 29 Dec 2019 13:42:22 INFO     Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Sun, 29 Dec 2019 13:42:23 INFO     Compiling out_2019-12-29/asm_tests/riscv_arithmetic_basic_test_0.S
riscv64-unknown-elf-gcc: error: out_2019-12-29/asm_tests/riscv_arithmetic_basic_test_0.S: No such file or directory
riscv64-unknown-elf-gcc: fatal error: no input files
compilation terminated.
Traceback (most recent call last):
  File "run.py", line 563, in <module>
    main()
  File "run.py", line 551, in main
    gcc_compile(matched_list, output_dir, args.isa, args.mabi, args.gcc_opts)
  File "run.py", line 282, in gcc_compile
    output = subprocess.check_output(cmd.split())
  File "/usr/lib64/python3.6/subprocess.py", line 356, in check_output
    **kwargs).stdout
  File "/usr/lib64/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/opt/riscv/bin/riscv64-unknown-elf-gcc', '-static', '-mcmodel=medany', '-fvisibility=hidden', '-nostdlib', '-nostartfiles', 'out_2019-12-29/asm_tests/riscv_arithmetic_basic_test_0.S', '-I/home/engr/riscv-dv/user_extension', '-T/home/engr/riscv-dv/scripts/link.ld', '-o', 'out_2019-12-29/asm_tests/riscv_arithmetic_basic_test_0.o', '-march=rv64gc', '-mabi=lp64']' returned non-zero exit status 1.

[engr@localhost riscv-dv]$ which riscv64-unknown-elf-gcc
/opt/riscv/bin/riscv64-unknown-elf-gcc
[engr@localhost riscv-dv]$ which riscv64-unknown-elf-objcopy 
/opt/riscv/bin/riscv64-unknown-elf-objcopy
[engr@localhost riscv-dv]$

danghai, 2019-12-29:
Could you run with `-v` option? I think it cannot generate the *.S assembly file, and it is not GCC issue.

taoliug, 2019-12-31:
yes, please run with -v and paste the log here.

weedmank, 2020-01-01:
kirk@KVM-Ubuntu19:~/riscv-dv/riscv-dv$ python3 -v run.py --test riscv_arithmetic_basic_test --simulator questa
import _frozen_importlib # frozen
import _imp # builtin
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import '_warnings' # <class '_frozen_importlib.BuiltinImporter'>
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
# installing zipimport hook
import 'zipimport' # <class '_frozen_importlib.BuiltinImporter'>
# installed zipimport hook
import '_frozen_importlib_external' # <class '_frozen_importlib.FrozenImporter'>
import '_io' # <class '_frozen_importlib.BuiltinImporter'>
import 'marshal' # <class '_frozen_importlib.BuiltinImporter'>
import 'posix' # <class '_frozen_importlib.BuiltinImporter'>
import _thread # previously loaded ('_thread')
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import _weakref # previously loaded ('_weakref')
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.7/encodings/__pycache__/__init__.cpython-37.pyc matches /usr/lib/python3.7/encodings/__init__.py
# code object from '/usr/lib/python3.7/encodings/__pycache__/__init__.cpython-37.pyc'
# /usr/lib/python3.7/__pycache__/codecs.cpython-37.pyc matches /usr/lib/python3.7/codecs.py
# code object from '/usr/lib/python3.7/__pycache__/codecs.cpython-37.pyc'
import '_codecs' # <class '_frozen_importlib.BuiltinImporter'>
import 'codecs' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb681fd0>
# /usr/lib/python3.7/encodings/__pycache__/aliases.cpython-37.pyc matches /usr/lib/python3.7/encodings/aliases.py
# code object from '/usr/lib/python3.7/encodings/__pycache__/aliases.cpython-37.pyc'
import 'encodings.aliases' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb6959e8>
import 'encodings' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb681a58>
# /usr/lib/python3.7/encodings/__pycache__/utf_8.cpython-37.pyc matches /usr/lib/python3.7/encodings/utf_8.py
# code object from '/usr/lib/python3.7/encodings/__pycache__/utf_8.cpython-37.pyc'
import 'encodings.utf_8' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb6a5710>
import '_signal' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.7/encodings/__pycache__/latin_1.cpython-37.pyc matches /usr/lib/python3.7/encodings/latin_1.py
# code object from '/usr/lib/python3.7/encodings/__pycache__/latin_1.cpython-37.pyc'
import 'encodings.latin_1' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb6a9208>
# /usr/lib/python3.7/__pycache__/io.cpython-37.pyc matches /usr/lib/python3.7/io.py
# code object from '/usr/lib/python3.7/__pycache__/io.cpython-37.pyc'
# /usr/lib/python3.7/__pycache__/abc.cpython-37.pyc matches /usr/lib/python3.7/abc.py
# code object from '/usr/lib/python3.7/__pycache__/abc.cpython-37.pyc'
import '_abc' # <class '_frozen_importlib.BuiltinImporter'>
import 'abc' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb6a97f0>
import 'io' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb6a9438>
# /usr/lib/python3.7/__pycache__/site.cpython-37.pyc matches /usr/lib/python3.7/site.py
# code object from '/usr/lib/python3.7/__pycache__/site.cpython-37.pyc'
# /usr/lib/python3.7/__pycache__/os.cpython-37.pyc matches /usr/lib/python3.7/os.py
# code object from '/usr/lib/python3.7/__pycache__/os.cpython-37.pyc'
# /usr/lib/python3.7/__pycache__/stat.cpython-37.pyc matches /usr/lib/python3.7/stat.py
# code object from '/usr/lib/python3.7/__pycache__/stat.cpython-37.pyc'
import '_stat' # <class '_frozen_importlib.BuiltinImporter'>
import 'stat' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb643b00>
# /usr/lib/python3.7/__pycache__/posixpath.cpython-37.pyc matches /usr/lib/python3.7/posixpath.py
# code object from '/usr/lib/python3.7/__pycache__/posixpath.cpython-37.pyc'
# /usr/lib/python3.7/__pycache__/genericpath.cpython-37.pyc matches /usr/lib/python3.7/genericpath.py
# code object from '/usr/lib/python3.7/__pycache__/genericpath.cpython-37.pyc'
import 'genericpath' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb64d5f8>
import 'posixpath' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb648278>
# /usr/lib/python3.7/__pycache__/_collections_abc.cpython-37.pyc matches /usr/lib/python3.7/_collections_abc.py
# code object from '/usr/lib/python3.7/__pycache__/_collections_abc.cpython-37.pyc'
import '_collections_abc' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb64dc50>
import 'os' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb6b57b8>
# /usr/lib/python3.7/__pycache__/_sitebuiltins.cpython-37.pyc matches /usr/lib/python3.7/_sitebuiltins.py
# code object from '/usr/lib/python3.7/__pycache__/_sitebuiltins.cpython-37.pyc'
import '_sitebuiltins' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb6b5b70>
# /usr/lib/python3.7/__pycache__/sitecustomize.cpython-37.pyc matches /usr/lib/python3.7/sitecustomize.py
# code object from '/usr/lib/python3.7/__pycache__/sitecustomize.cpython-37.pyc'
# /usr/lib/python3/dist-packages/__pycache__/apport_python_hook.cpython-37.pyc matches /usr/lib/python3/dist-packages/apport_python_hook.py
# code object from '/usr/lib/python3/dist-packages/__pycache__/apport_python_hook.cpython-37.pyc'
import 'apport_python_hook' # <_frozen_importlib_external.SourceFileLoader object at 0x7fa0cb5e5940>
[... cut ...]
# destroy yaml.dumper
# destroy yaml.cyaml
# cleanup[3] wiping base64
# destroy struct
# cleanup[3] wiping _struct
# destroy _struct
# cleanup[3] wiping binascii
# cleanup[3] wiping importlib._bootstrap
# cleanup[3] wiping sys
# cleanup[3] wiping builtins
# destroy binascii
# destroy datetime
# destroy copyreg
# destroy types
# destroy base64
# destroy stat
# destroy genericpath
# destroy functools
# destroy tokenize
# destroy _heapq
# destroy _sre
# destroy sre_parse
kirk@KVM-Ubuntu19:~/riscv-dv/riscv-dv$

weedmank, 2020-01-01:
Looks like '# ' at the beginning of lines caused the large print above.

taoliug, 2020-01-01:
Can you highlight the error message?

weedmank, 2020-01-01:
OK.  Only the error section shows in large print now. It's about 1/3 of the way down the report

taoliug, 2020-01-01:
Can you share sim_riscv_arithmetic_basic_test.log under output directory?

weedmank, 2020-01-01:
I don't see any files.

kirk@KVM-Ubuntu19:~/Desktop$ cd
kirk@KVM-Ubuntu19:~$ cd riscv-dv/riscv-dv/
kirk@KVM-Ubuntu19:~/riscv-dv/riscv-dv$ ls
CONTRIBUTING.md  LICENSE.txt     README.md  setting  test                  yaml
cov.py           out_2019-12-31  run.py     src      user_extension
files.f          questa_sim.tcl  scripts    target   vcs.compile.option.f
kirk@KVM-Ubuntu19:~/riscv-dv/riscv-dv$ cd out_2019-12-31/
kirk@KVM-Ubuntu19:~/riscv-dv/riscv-dv/out_2019-12-31$ ls
asm_tests  seed.yaml
kirk@KVM-Ubuntu19:~/riscv-dv/riscv-dv/out_2019-12-31$ cd asm_tests/
kirk@KVM-Ubuntu19:~/riscv-dv/riscv-dv/out_2019-12-31/asm_tests$ ls
kirk@KVM-Ubuntu19:~/riscv-dv/riscv-dv/out_2019-12-31/asm_tests

weedmank, 2020-01-01:
> I don't see any files.
>
> kirk@KVM-Ubuntu19:~/Desktop$ cd kirk@KVM-Ubuntu19:~$ cd riscv-dv/riscv-dv/
> kirk@KVM-Ubuntu19:~/riscv-dv/riscv-dv$ ls CONTRIBUTING.md LICENSE.txt README.md setting test yaml [...]

closed 2020-01-01 by weedmank

taoliug, 2020-01-01:
put the log in a code block "<>"  might render better.
BTW, I think the simulation doesn't run, might be something related to questa environment setup.

taoliug, 2020-01-01:
Can you try "python3 run.py -v --test riscv_arithmetic_basic_test --simulator questa"?

weedmank, 2020-01-01:
I thin I know what the questa problem is.  I'll work on it. But is that really causing run.py not to create the .s file?

kirk@KVM-Ubuntu19:~/riscv-dv/riscv-dv$ python3 run.py -v --test riscv_arithmetic_basic_test --simulator questa
Wed, 01 Jan 2020 11:32:18 lib.py:174   INFO     Processing regression test list : /home/kirk/riscv-dv/riscv-dv/yaml/testlist.yaml, test: riscv_arithmetic_basic_test
Wed, 01 Jan 2020 11:32:18 lib.py:182   INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:2
Wed, 01 Jan 2020 11:32:18 run.py:47    INFO     Processing simulator setup file : /home/kirk/riscv-dv/riscv-dv/yaml/simulator.yaml
Wed, 01 Jan 2020 11:32:18 run.py:52    INFO     Found matching simulator: questa
Wed, 01 Jan 2020 11:32:18 run.py:161   INFO     Building RISC-V instruction generator
Wed, 01 Jan 2020 11:32:18 run.py:172   DEBUG    Compile command: vmap mtiUvm $QUESTA_HOME/questasim/uvm-1.2
Wed, 01 Jan 2020 11:32:18 lib.py:111   DEBUG    vmap mtiUvm $QUESTA_HOME/questasim/uvm-1.2
Wed, 01 Jan 2020 11:32:18 lib.py:128   DEBUG    /bin/bash: line 0: exec: vmap: not found

Wed, 01 Jan 2020 11:32:18 run.py:173   DEBUG    /bin/bash: line 0: exec: vmap: not found

Wed, 01 Jan 2020 11:32:18 run.py:172   DEBUG    Compile command: vlog -64 +incdir+/home/kirk/riscv-dv/riscv-dv/setting +incdir+/home/kirk/riscv-dv/riscv-dv/user_extension -access=rwc -f /home/kirk/riscv-dv/riscv-dv/files.f -sv -mfcu -cuname design_cuname +define+UVM_REGEX_NO_DPI -writetoplevels /home/kirk/riscv-dv/riscv-dv/out_2020-01-01/top.list -l /home/kirk/riscv-dv/riscv-dv/out_2020-01-01/compile.log 
Wed, 01 Jan 2020 11:32:18 lib.py:111   DEBUG    vlog -64 +incdir+/home/kirk/riscv-dv/riscv-dv/setting +incdir+/home/kirk/riscv-dv/riscv-dv/user_extension -access=rwc -f /home/kirk/riscv-dv/riscv-dv/files.f -sv -mfcu -cuname design_cuname +define+UVM_REGEX_NO_DPI -writetoplevels /home/kirk/riscv-dv/riscv-dv/out_2020-01-01/top.list -l /home/kirk/riscv-dv/riscv-dv/out_2020-01-01/compile.log 
Wed, 01 Jan 2020 11:32:18 lib.py:128   DEBUG    /bin/bash: line 0: exec: vlog: not found

Wed, 01 Jan 2020 11:32:18 run.py:173   DEBUG    /bin/bash: line 0: exec: vlog: not found

Wed, 01 Jan 2020 11:32:18 run.py:172   DEBUG    Compile command: vopt -64 -debug +designfile -f /home/kirk/riscv-dv/riscv-dv/out_2020-01-01/top.list -l /home/kirk/riscv-dv/riscv-dv/out_2020-01-01/optimize.log  -o design_opt
Wed, 01 Jan 2020 11:32:18 lib.py:111   DEBUG    vopt -64 -debug +designfile -f /home/kirk/riscv-dv/riscv-dv/out_2020-01-01/top.list -l /home/kirk/riscv-dv/riscv-dv/out_2020-01-01/optimize.log  -o design_opt
Wed, 01 Jan 2020 11:32:18 lib.py:128   DEBUG    /bin/bash: line 0: exec: vopt: not found

Wed, 01 Jan 2020 11:32:18 run.py:173   DEBUG    /bin/bash: line 0: exec: vopt: not found

Wed, 01 Jan 2020 11:32:18 run.py:184   INFO     Running RISC-V instruction generator
Wed, 01 Jan 2020 11:32:18 run.py:188   INFO     Generating 2 riscv_arithmetic_basic_test
Wed, 01 Jan 2020 11:32:18 run.py:210   INFO     Running riscv_arithmetic_basic_test with 1 batches
Wed, 01 Jan 2020 11:32:18 run.py:237   INFO     Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Wed, 01 Jan 2020 11:32:18 lib.py:111   DEBUG     vsim -64 -c  -do /home/kirk/riscv-dv/riscv-dv/questa_sim.tcl design_opt   -sv_seed 4281266173 +UVM_TESTNAME=riscv_instr_base_test  +num_of_tests=2  +start_idx=0  +asm_file_name=out_2020-01-01/asm_tests/riscv_arithmetic_basic_test  -l out_2020-01-01/sim_riscv_arithmetic_basic_test_0.log +instr_cnt=10000 +num_of_sub_program=0 +no_fence=1 +no_data_page=1 +no_branch_jump=1 +boot_mode=m +no_csr_instr=1

Wed, 01 Jan 2020 11:32:18 lib.py:128   DEBUG    /bin/bash: line 0: exec: vsim: not found

Wed, 01 Jan 2020 11:32:18 run.py:284   INFO     Compiling out_2020-01-01/asm_tests/riscv_arithmetic_basic_test_0.S
Wed, 01 Jan 2020 11:32:18 run.py:285   DEBUG    /opt/riscv/bin/riscv64-unknown-elf-gcc -static -mcmodel=medany              -fvisibility=hidden -nostdlib              -nostartfiles out_2020-01-01/asm_tests/riscv_arithmetic_basic_test_0.S              -I./user_extension              -T./scripts/link.ld  -o out_2020-01-01/asm_tests/riscv_arithmetic_basic_test_0.o  -march=rv64gc -mabi=lp64
riscv64-unknown-elf-gcc: error: out_2020-01-01/asm_tests/riscv_arithmetic_basic_test_0.S: No such file or directory
riscv64-unknown-elf-gcc: fatal error: no input files
compilation terminated.
Traceback (most recent call last):
  File "run.py", line 572, in <module>
    main()
  File "run.py", line 560, in main
    gcc_compile(matched_list, output_dir, args.isa, args.mabi, args.gcc_opts)
  File "run.py", line 286, in gcc_compile
    output = subprocess.check_output(cmd.split())
  File "/usr/lib/python3.7/subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.7/subprocess.py", line 487, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/opt/riscv/bin/riscv64-unknown-elf-gcc', '-static', '-mcmodel=medany', '-fvisibility=hidden', '-nostdlib', '-nostartfiles', 'out_2020-01-01/asm_tests/riscv_arithmetic_basic_test_0.S', '-I./user_extension', '-T./scripts/link.ld', '-o', 'out_2020-01-01/asm_tests/riscv_arithmetic_basic_test_0.o', '-march=rv64gc', '-mabi=lp64']' returned non-zero exit status 1.
kirk@KVM-Ubuntu19:~/riscv-dv/riscv-dv$

taoliug, 2020-01-02:
I think this is the problem:
Wed, 01 Jan 2020 11:32:18 lib.py:128 DEBUG /bin/bash: line 0: exec: vmap: not found
Wed, 01 Jan 2020 11:32:18 run.py:173 DEBUG /bin/bash: line 0: exec: vopt: not found
Wed, 01 Jan 2020 11:32:18 lib.py:128 DEBUG /bin/bash: line 0: exec: vsim: not found

Can you check if your questa sim environment is setup properly?