chipsalliance/riscv-dv #399: Getting a lot of warning for riscv_arithmetic_basic_test

danghai, 2019-12-30:
I get a lot of warning for this test `riscv_arithmetic_basic_test`.
```
ai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 219
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff1a8963 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 219
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff1a8963 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 219
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff1a8963 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 219
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff1a8963 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 219
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff1a8963 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 219
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff1a8963 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 219
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff1a8963 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 219
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff1a8963 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 219
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff1a8963 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 219
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff1a8963 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 219
```

abdullahyildiz, 2019-12-30:
@danghai,

Which tool (along with its version) are you using?

danghai, 2019-12-30:
I use questa, and version 2019.4

abdullahyildiz, 2019-12-30:
I don't remember experiencing such an issue when I try it on Synopsys VCS. Let me look at the verbose log file of the same case when VCS has been run.

danghai, 2019-12-31:
It still has warning.
```
sa/riscv_instr.sv Line: 221
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff028c10 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 221
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff028c10 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 221
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff028c10 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 221
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff028c10 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 221
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff028c10 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 221
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff028c10 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 221
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff028c10 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 221
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff028c10 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 221
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff028c10 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 221
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff028c10 File: /home/danghai/hhoangda/riscv-dv/src/isa/riscv_instr.sv Line: 221
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
```

taoliug, 2019-12-31:
please verify if it fixes the issue

danghai, 2019-12-31:
Yes, It is fixed.

PR #400 (2019-12-31): Fix simulation warning
fix #399
closed 2019-12-31 by danghai