chipsalliance/riscv-dv #403: Support vector extension

taoliug, 2019-12-31:
I have started working on adding vector extension support based on riscv vector extension spec v0.8. File this bug to track implementation progress. Please leave your comment if you have any question about this.

taoliug, 2020-01-02:
The basic vector arithmetic test is ready to run now. 

Toolchain
gcc: https://github.com/riscv/riscv-gnu-toolchain/tree/rvv-0.8.x
spike: https://github.com/riscv/riscv-isa-sim/tree/b8806095104da9c41e65696840c2863a08f82936
ovpsim: https://github.com/riscv/riscv-ovpsim

command:
python3 run.py -i 1 --target rv64gcv -tn riscv_vector_arithmetic_test

Known issues:
- [Spike doesn't support vector instruction commit log at this point](https://github.com/riscv/riscv-isa-sim/issues/374)
- [Some ovpsim vector instruction decoding issue under debugging](https://github.com/riscv/riscv-ovpsim/issues/34)

taoliug, 2020-01-04:
All instructions are added except for vector load/store. Simulation can pass with ovpsim. 

A few more things need to be added:
- Add vector load/store instructions
- Randomize vtype/vl, enable random mask
- Add functional coverage model for vector instructions
- Trace log process script update for vector instructions.

Sustrak, 2020-02-21:
Hi do you know when the rest of the vector instructions will be implemented ?

Thanks for your effort.

Sustrak, 2020-04-02:
Hi,
for the project I'm working on I need to be able to generate memory operation instructions. Will those be implemented in the near future ? If not, do you have any guidelines on how you would like to implement them ? I can do the effort.

Thanks

taoliug, 2020-04-24:
We expect to resume this development in a couple of weeks.

taoliug, 2020-06-03:
We are currently integrating the Vector extension work contributed by Andes, expect to push the changes in the next few days.

taoliug, 2020-06-07:
Randomized LMUL and VSEW is now supported.
Following work
- Integrate Andes's vector load/store instruction stream to upstream

Below features are not supported yet:
- Zvediv extension
- Zvqmac extension

taoliug, 2020-08-19:
Closing this bug now as most of the v0.8 features are implemented. A tag has been created from vector extension v0.8 support. 
https://github.com/google/riscv-dv/releases/tag/v0.8

We are moving on to upgrade the upstream to support v0.9 vector extension spec

closed 2020-08-19 by taoliug