chipsalliance/riscv-dv #406: It has warning for running coverage

danghai, 2019-12-31:
It has warning for running coverage. Step to reproduce:
```
python3 run.py --test riscv_arithmetic_basic_test --iss spike --simulator qrun -o qruncov -v
python3 cov.py --dir qruncov/spike_sim --simulator qrun -v
```

Log file:
```
 UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] pc=ffffffff8000e0dc
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] instr=beq
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] gpr=
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] csr=
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] binary=36fb8763
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] mode=
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] instr_str=beq     s7, a5, 878
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] operand=s7,a5,878
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff02ee09 File: /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv Line: 172
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(80) @ 0: uvm_test_top [] ----------------------------------------------------------
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] pc=ffffffff8000e0e0
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] instr=c.li
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] gpr=a5:00000002
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] csr=
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] binary=00004789
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] mode=3
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] instr_str=c.li    a5, 2
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [] operand=a5,2
# ** Warning: (vsim-3829) Non-existent associative array entry. Returning default value.
#    Time: 0 ns  Iteration: 61  Process: /uvm_pkg::uvm_task_phase::execute/#FORK#143(#ublk#215181159#143)_7feff02ee09 File: /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv Line: 172
# UVM_INFO /home/danghai/hhoangda/riscv-dv/test/riscv_instr_cov_test.sv(80) @ 0: uvm_test_top [] ------------------------------------
```

taoliug, 2020-01-01:
please verify if it's fixed.

danghai, 2020-01-01:
It works now

closed 2020-01-01 by danghai