chipsalliance/riscv-dv #411: Multi-thread feature request for riscv-dv

vandanaprabhu, 2020-01-01:
Hi,
Is there a proposal/plan to implement a feature that allows running google riscv-dv tests for multiple threads (two to begin with)? 
Thanks,
Vandana

taoliug, 2020-01-01:
If you have LSF, you can use the following command to make the generator run multiple tests in parallel.
`python3 run.py --lsf_cmd "bsub -Is" ....`

Multi-thread on a single machine is not supported.

taoliug, 2020-01-01:
BTW,  here's the function for multi-thread execution.  It will only be invoked when you pass lsf_cmd in the command line. 
https://github.com/google/riscv-dv/blob/master/scripts/lib.py#L135

vandanaprabhu, 2020-01-01:
Sorry, that's not what I meant. I was enquiring about support for generating random tests on multiple harts, to the extent where each generated test will include the initial code for checking if it's running on a specific thread and jump to a different starting address based on the thread it's configured for and based on the assumption that there will be no shared memory accesses during the execution of multiple harts to begin with, and possibly at a later stage, add support for generating tests for multiple harts that share memory space.

taoliug, 2020-01-02:
Got it, this is not in the near term plan. We are unlikely to address it in 2020 H1, but I could put it in the future plan. If you have any proposal, please let me know. I can evaluate the complexity of adding this feature.

danghai, 2020-01-03:
@taoliug . for `lsf_cmd`. I am not quite sure why It does not work for me. Ex:
`python3 run.py --lsf_cmd="bsub -Is" --simulator qrun -tn riscv_arithmetic_basic_test`
The output:
```
Fri, 03 Jan 2020 13:16:44 INFO     Creating output directory: out_2020-01-03
Fri, 03 Jan 2020 13:16:44 INFO     Processing regression test list : /home/danghai/hhoangda/riscv-dv/target/rv32imc/testlist.yaml, test: riscv_arithmetic_basic_test
Fri, 03 Jan 2020 13:16:44 INFO     Processing regression test list : /home/danghai/hhoangda/riscv-dv/yaml/base_testlist.yaml, test: riscv_arithmetic_basic_test
Fri, 03 Jan 2020 13:16:44 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:2
Fri, 03 Jan 2020 13:16:44 INFO     Processing simulator setup file : /home/danghai/hhoangda/riscv-dv/yaml/simulator.yaml
Fri, 03 Jan 2020 13:16:44 INFO     Found matching simulator: qrun
Fri, 03 Jan 2020 13:16:44 INFO     Building RISC-V instruction generator
Fri, 03 Jan 2020 13:16:50 INFO     Running RISC-V instruction generator
Fri, 03 Jan 2020 13:16:50 INFO     Generating 2 riscv_arithmetic_basic_test
Fri, 03 Jan 2020 13:16:50 INFO     Running riscv_arithmetic_basic_test with 1 batches
Fri, 03 Jan 2020 13:16:50 INFO     Command progress: 0/1
Fri, 03 Jan 2020 13:16:50 INFO     /bin/bash: line 0: exec: bsub: not found

Fri, 03 Jan 2020 13:16:50 ERROR    ERROR return code: 127, cmd:bsub -Is qrun -64 -simulate -snapshot design_opt -c   -sv_seed 157562186 -outdir /home/danghai/hhoangda/riscv-dv/out_2020-01-03/qrun.out +UVM_TESTNAME=riscv_instr_base_test  +num_of_tests=2  +start_idx=0  +asm_file_name=out_2020-01-03/asm_tests/riscv_arithmetic_basic_test  -l out_2020-01-03/sim_riscv_arithmetic_basic_test_0.log +instr_cnt=10000 +num_of_sub_program=0 +directed_instr_0=riscv_int_numeric_corner_stream,4 +no_fence=1 +no_data_page=1 +no_branch_jump=1 +boot_mode=m +no_csr_instr=1

Fri, 03 Jan 2020 13:16:50 ERROR    Cannot find assembly test: out_2020-01-03/asm_tests/riscv_arithmetic_basic_test_0.S

```

taoliug, 2020-01-03:
I think most likely it's because the LSF machine cannot access your home directory.

taoliug, 2020-01-03:
try to use -o to specify a full output directory path (do not use default relative path). BTW, please file separate issue for this.

taoliug, 2020-01-03:
I am closing this issue for now as there's no plan to address it in the short term. I will file a separate issue once we get to it.

closed 2020-01-03 by taoliug

ganoam, 2021-04-27:
> @taoliug . for `lsf_cmd`. I am not quite sure why It does not work for me. Ex:
> `python3 run.py --lsf_cmd="bsub -Is" --simulator qrun -tn riscv_arithmetic_basic_test`
> The output:
>
> ```
> Fri, 03 Jan [...]
Sorry to post in this closed issue... I got the same error. Just wanted to quickly ask if you found the reason for it, @danghai?

ganoam, 2021-04-27:
> Sorry to post in this closed issue... I got the same error. Just wanted to quickly ask if you found the reason for it, @danghai?

Edit: Nevermind, got it...