chipsalliance/riscv-dv #417: Issue relating to running multiple generator jobs in parallel through LSF

danghai, 2020-01-03:
Command to produce:
`python3 run.py --lsf_cmd="bsub -Is" --simulator qrun -tn riscv_arithmetic_basic_test -o lfs_test`
The log file:
```
Fri, 03 Jan 2020 13:23:37 INFO     Creating output directory: lfs_test
Fri, 03 Jan 2020 13:23:37 INFO     Processing regression test list : /home/danghai/hhoangda/riscv-dv/target/rv32imc/testlist.yaml, test: riscv_arithmetic_basic_test
Fri, 03 Jan 2020 13:23:37 INFO     Processing regression test list : /home/danghai/hhoangda/riscv-dv/yaml/base_testlist.yaml, test: riscv_arithmetic_basic_test
Fri, 03 Jan 2020 13:23:37 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:2
Fri, 03 Jan 2020 13:23:37 INFO     Processing simulator setup file : /home/danghai/hhoangda/riscv-dv/yaml/simulator.yaml
Fri, 03 Jan 2020 13:23:37 INFO     Found matching simulator: qrun
Fri, 03 Jan 2020 13:23:37 INFO     Building RISC-V instruction generator
Fri, 03 Jan 2020 13:23:42 INFO     Running RISC-V instruction generator
Fri, 03 Jan 2020 13:23:42 INFO     Generating 2 riscv_arithmetic_basic_test
Fri, 03 Jan 2020 13:23:42 INFO     Running riscv_arithmetic_basic_test with 1 batches
Fri, 03 Jan 2020 13:23:42 INFO     Command progress: 0/1
Fri, 03 Jan 2020 13:23:42 INFO     /bin/bash: line 0: exec: bsub: not found

Fri, 03 Jan 2020 13:23:42 ERROR    ERROR return code: 127, cmd:bsub -Is qrun -64 -simulate -snapshot design_opt -c   -sv_seed 1297782651 -outdir /home/danghai/hhoangda/riscv-dv/lfs_test/qrun.out +UVM_TESTNAME=riscv_instr_base_test  +num_of_tests=2  +start_idx=0  +asm_file_name=lfs_test/asm_tests/riscv_arithmetic_basic_test  -l lfs_test/sim_riscv_arithmetic_basic_test_0.log +instr_cnt=10000 +num_of_sub_program=0 +directed_instr_0=riscv_int_numeric_corner_stream,4 +no_fence=1 +no_data_page=1 +no_branch_jump=1 +boot_mode=m +no_csr_instr=1

Fri, 03 Jan 2020 13:23:42 ERROR    Cannot find assembly test: lfs_test/asm_tests/riscv_arithmetic_basic_test_0.S

```

taoliug, 2020-01-03:
It really depends on how your LSF environment is setup. --lsf_cmd = "bsub -Is" is just an example if you use bsub to submit jobs.

danghai, 2020-01-03:
I do not see it mentions setting LSF environment in README

danghai, 2020-01-03:
Do you mean this one: https://www.ibm.com/support/knowledgecenter/en/SSETD4_9.1.2/lsf_command_ref/bsub.1.html ?

taoliug, 2020-01-03:
Yes, it really depends on your job system.

closed 2020-01-04 by taoliug