chipsalliance/riscv-dv #418: Python package

danghai, 2020-01-04:
The project riscv-dv should have a python package? And then, it can be public on PyPI for easier download by pip?

taoliug, 2020-01-04:
taliu@taliu0:~/riscv-dv$ pip3 install -e .
Requirement already satisfied: PyYAML in /usr/lib/python3/dist-packages (from riscv-dv==1.0) (5.1.2)
Requirement already satisfied: bitstring in ~/.local/lib/python3.7/site-packages (from riscv-dv==1.0) (3.1.6)
Installing collected packages: riscv-dv
  Found existing installation: riscv-dv 1.0
    Uninstalling riscv-dv-1.0:
      Successfully uninstalled riscv-dv-1.0
  Running setup.py develop for riscv-dv
Successfully installed riscv-dv
taliu@taliu0:~/riscv-dv$ run
**bash: run: command not found**

Seems like i cannot run it, PTAL.

danghai, 2020-01-04:
Could you try `pip3 install --user git+https://github.com/google/riscv-dv.git`?

danghai, 2020-01-04:
It works for me for both ways. Could you clean/remove your directory, and re-try it?

taoliug, 2020-01-04:
It shows the installation is successful, but no local run/cov executable is created under riscv-dv

danghai, 2020-01-04:
Could you try uninstall, and then reinstall again:
```
pip3 uninstall -y riscv-dv
pip3 install --user -e .
```

For me it shows:
```
danghai@ubuntu:~/hhoangda/riscv-dv$ which cov
/home/danghai/.local/bin/cov
```

taoliug, 2020-01-04:
I can see it's installed under ~/.local/bin, but the ~/.local/bin is not in the $PATH environment variable, so it cannot be found automatically.

taoliug, 2020-01-04:
Solved the problem by adding ~/.local/bin to the local variable. Maybe we need to mention this in the README, it might be a common problem for ubuntu
https://askubuntu.com/questions/799302/ubuntu-cant-find-an-executable-file-in-local-bin

danghai, 2020-01-04:
Cool! I will mention it. You can run it outside the `riscv-dv` directory that you cloned? I mean you can run it anywhere without the hardware source? I put all files into MANIFEST. But I cannot find the location of package.

taoliug, 2020-01-04:
One problem is that if I have multiple clones, how can I run for each clone? Should the python package be put under riscv-dv directory?

taoliug, 2020-01-04:
I think this might be OK, as I can always run with "python3 xxx/xxx/run.py ..."

danghai, 2020-01-04:
For multiple clones, you should re-install it by `pip3 install --user -e .` for each clone.

taoliug, 2020-01-05:
What if I want to work on two clones in parallel? I believe there will just be one copy of package under .local/bin?

danghai, 2020-01-05:
For this situation, I think you should use a virtual environment. For example:
```
virtualenv ~/riscv-venv/
source ~/riscv-venv/bin/activate
pip install --user -e .
```
And deactive when you done: `deactivate`.
it keeps all its dependencies in a separate Python environment.

taoliug, 2020-01-05:
I think it's more complicated than the current use model. In many cases, I may have multiple local clones working on different things. We should recommend user to use the below approach in this scenario.

python3 <path1>/run.py ....
python3 <path2>/run.py ....

I am also a bit worry about the naming conflict. "run" seems to be too general which could collide with other executables. Anyway, the new use model does make things easier in many cases, but we should clarify the scenarios that the old use model is recommended.

taoliug, 2020-01-05:
I think for developers which may work on multiple clones in parallel, we should recommend to use the old flow. For normal users which just has one clone, the python package approach is recommended.

danghai, 2020-01-05:
Okay I will mention it. All examples use the old flow, or python package?

taoliug, 2020-01-05:
I think using python package is fine, just have a separate section to call out the use model for the developers.

commit d4eec438 (2020-01-04): Update information about instruction for running scripts (#420)

* riscv-dv requires python >=3 for install

* README.md: Update information about running script

Fixed #418
closed 2020-01-05 by danghai