chipsalliance/riscv-dv #425: Change documentation format

taoliug, 2020-01-05:
Right now the main document of riscv-dv is README.md. As more content is added, it becomes too long to read in one page. The plan is to divide it to small chapters and use .rst format instead of .md, then use Sphinx to generate the complete document. 
Free feel to let me know if you want to contribute to this effort, this will enable us to provide better documentation to users.

taoliug, 2020-01-05:
Here's the planned structure of the document:

- Overview
- Getting started (including environment setup and basic commands)
- Configuration
- End to end simulation flow
- Generator flow [empty now]
- Coverage model
- Customize and extend the generator
- Class reference [empty now]
- Command line reference [empty now]

taoliug, 2020-01-06:
Thanks @danghai to get it done. I think the remaining work is to generate PDF or HTML document under docs directory, and remove the duplicated information in the README.md (add a pointer to the new doc)

danghai, 2020-01-07:
You mean copy all html code into upsteam?

taoliug, 2020-01-07:
Maybe PDF is better as it's just one file?

danghai, 2020-01-07:
I do not see pdf file in make:
```
  html        to make standalone HTML files
  dirhtml     to make HTML files named index.html in directories
  singlehtml  to make a single large HTML file
  pickle      to make pickle files
  json        to make JSON files
  htmlhelp    to make HTML files and an HTML help project
  qthelp      to make HTML files and a qthelp project
  devhelp     to make HTML files and a Devhelp project
  epub        to make an epub
  latex       to make LaTeX files, you can set PAPER=a4 or PAPER=letter
  latexpdf    to make LaTeX and PDF files (default pdflatex)
  latexpdfja  to make LaTeX files and run them through platex/dvipdfmx
  text        to make text files
  man         to make manual pages
  texinfo     to make Texinfo files
  info        to make Texinfo files and run them through makeinfo
  gettext     to make PO message catalogs
  changes     to make an overview of all changed/added/deprecated items
  xml         to make Docutils-native XML files
  pseudoxml   to make pseudoxml-XML files for display purposes
  linkcheck   to check all external links for integrity
  doctest     to run all doctests embedded in the documentation (if enabled)
  coverage    to run coverage check of the documentation (if enabled)
```
I try latexpdf, but it create a lot of file that I am not familiar with

danghai, 2020-01-07:
Ah I can generate 1 file pdf

closed 2020-01-11 by taoliug