chipsalliance/riscv-dv #440: Support C test

taoliug, 2020-01-14:
Similar to handcoded assembly test support, it will be great to allow user to write C test as a supplement approach to test the processor.  Example:

```yaml
- test: riscv_directed_c_test
  description: >
    Directed C test
  gcc_opts: >
     xxxxx
  c_test : <C test file name or test directory name>
  iterations: 1
```

danghai, 2020-01-15:
Hi @taoliug . I try to create a simple C file. And then compiling by `riscv64-unknown-elf-gcc`
```
/opt/riscv/bin/riscv64-unknown-elf-gcc -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles test.c -I/home/danghai/hhoangda/riscv-dv/user_extension -T/home/danghai/hhoangda/riscv-dv/scripts/link.ld -o c_test.o 
/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 0000000080000000
```
I am not sure about the warning: `cannot find entry symbol _start; defaulting to 0000000080000000`.
I use spike: `/opt/riscv/bin/spike --log-commits -l c_test.o`
```
core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
core   0:           tval 0x0000000000000000
core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
core   0:           tval 0x0000000000000000
core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
core   0:           tval 0x0000000000000000
core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
core   0:           tval 0x0000000000000000
core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
core   0:           tval 0x0000000000000000
core   0: exception trap_instruction_access_fault, epc 0x0000000000000000
```
Do you know to resolve it?

abdullahyildiz, 2020-01-15:
@danghai,

It seems that there is no crt.S file in compilation command.

You can take a look at the Makefile log by building standard RISC-V benchmarks to see how they use crt.S file which includes `_start` symbol..

taoliug, 2020-01-21:
@danghai, feel tree to submit what you have right now. I can try to fix this issue.

danghai, 2020-01-23:
Suggest to add a new field c_tests to allow user specify directed c test in the testlist. This will enable user to run random instruction test and directed c test in one regression.

Sample test entries
```
- test: riscv_single_c_test
  description: >
     single c test entry
  iterations: 1
  c_test: sample_c.c

- test: riscv_c_regression_test
  description: >
    Run all c tests under the given directory
  iterations: 1
  c_test: c_test_directory
  gcc_opts:
     # Some custom gcc options
```

udinator, 2020-04-17:
I believe this feature has been added a while ago, I'll go ahead and close this issue.

closed 2020-04-17 by udinator