chipsalliance/riscv-dv #443: Add coverage model for floating point extension

taoliug, 2020-01-17:
File this bug to track the implementation progress of floating point coverage model. A few things to start with:
- Cover all possible operands for each floating point instruction
- Numeric corner cases for the arithmetic instructions (NaN, subnormal values etc)
- Floating point CSR coverage (fcsr)

taoliug, 2020-05-29:
Weicai, can you give a summary what's left to be done for this task? Thanks

weicaiyang, 2020-05-29:
> Weicai, can you give a summary what's left to be done for this task? Thanks

@taoliug 
These are done
1. single precision 11.1-11.6 sections
2. parsing instruction should work for all floating point instructions

Remaining:
1. 11.7-11.9
2. chapter 12 double precision
3. floating point special values

If you're busy on integrating vector extension, I continue to this one.

weicaiyang, 2020-06-17:
with #615 merged, use more meaningful initial value for floating point registers and I see most of the floating point coverage are 100%. 
For instructions like FMADD, rd=rs1×rs2+rs3, it's hard to cover all kinds of special values for rd. Perhaps we need some direct tests or just remove the coverage point for rd, since we have cover all the other operands.

taoliug, 2020-07-04:
Closing this bug as all related work is completed

closed 2020-07-04 by taoliug