chipsalliance/riscv-dv #445: bit manipulation implementation in repo asher-ben-shooshan/mriscv

asher-ben-shooshan, 2020-01-20:
hi
in the repository asher-ben-shooshan/mriscv there is a description of the modifications
in order to implement the bit manipulation implementation and also in that repo there are
the needed new files that should be uploaded.
for your care

taoliug, 2020-01-20:
Thanks for your contribution, the code looks good to me. Do you want to create a PR to make it upstream?

asher-ben-shooshan, 2020-01-22:
בס"ד
Hi
Can you explain what is a “PR to make it upstream”?
I want to contribute it

Thanks
Asher

taoliug, 2020-01-22:
Just create a Pull Request to push your changes from your local branch to the upstream.
Try this: 
1) git push origin HEAD -f
2) Create a PR here : https://github.com/google/riscv-dv/pulls

asher-ben-shooshan, 2020-01-22:
בס"ד
Hi
Can you give me a phone number so I can call you since that I cant manage to create a PR
Thanks

taoliug, 2020-01-22:
Can you check if this information helpful?
https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request

taoliug, 2020-01-22:
I think this is what you need to do:
1. git clone https://github.com/google/riscv-dv
2. cd riscv-dv
3. git checkout -b dev
4. Copy your code from your own repo to this branch
5. git add .
6. git commit -m "Add bit-manipulation extension"
7. git push origin HEAD -f
8. Create a pull request

asher-ben-shooshan, 2020-01-22:
בס"ד
Hi
First thanks for your support
But I am failing on step 7 since I don’t have the correct username and pwd
Thanks

taoliug, 2020-01-22:
Can you post the error message here? Maybe you can get some help from other contributors as well.

asher-ben-shooshan, 2020-01-22:
בס"ד
Hi
Thanks

[cid:image001.png@01D5D102.59A34630]

[cid:image002.png@01D5D102.88CBE020]

It seems that I cant push to this branch since I don’t have the permissions.

Thanks

taoliug, 2020-01-22:
You might need to add a new remote, please check this
https://help.github.com/en/github/using-git/adding-a-remote

asher-ben-shooshan, 2020-01-22:
בס"ד
Hi
I really appreciate  the support
But it is not working maybe a I supposed to open a repo of my own and to make a pull request from there?
Thanks

https://help.github.com/en/github/using-git/adding-a-remote<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fhelp.github.com%2Fen%2Fgithub%2Fusing-git%2Fadding-a-remote&data=02%7C01%7Casher%40mellanox.com%7C68bfe00b63e24b4cf73b08d79f0b90cf%7Ca652971c7d2e4d9ba6a4d149256f461b%7C0%7C0%7C637152744333137646&sdata=j%2BIN8NkoxysbTAWWwDOCqjJ%2F2kWHs406sHGPy98SMUM%3D&reserved=0>

taoliug, 2020-01-22:
No, the current issue is your remote setup
Can you run "git remote -v" to get the current remote name

asher-ben-shooshan, 2020-01-22:
בס"ד
Hi
[cid:image001.png@01D5D107.7D769F80]

Is it good?
Thanks

asher-ben-shooshan, 2020-01-22:
בס"ד
Hi
The error response is:
[cid:image001.png@01D5D107.D1124770]

Thanks

taoliug, 2020-01-22:
I cannot see any image you posted.

asher-ben-shooshan, 2020-01-22:
בס"ד
Hi
I still did not succeed in sending it
It is failing
With error
[cid:image001.png@01D5D10D.C9524610]

Do you have any suggestions?

Thanks

danghai, 2020-01-22:
at step 1. git clone https://github.com/google/riscv-dv
`It seems that I cant push to this branch since I don’t have the permissions.`
He should fork, and clone his fork.
You can see `fork` at up right corner. Click the fork, and then clone your fork. For my fork
`https://github.com/danghai/riscv-dv.git`

asher-ben-shooshan, 2020-01-22:
בס"ד
Hi
Did
You get the files?
I think I managed to do the pull request

Thanks

asher-ben-shooshan, 2020-01-22:
בס"ד
Hi
After speaking with my company it was found out that I had a mistake
In signing the agreement that I have signed againt google
I have signed it as by yourself and it is needed that my company Mellanox will
Sign it instead of me.
So hereby I am cancelling the agreement I have signed today with Google.
Please confirm reading this message and cancelling the agreement.

Thanks
Asher

taoliug, 2020-01-27:
Please let me know your plan going forward, thanks.

asher-ben-shooshan, 2020-01-29:
בס"ד
Hi
Thanks a lot
I will let you know when I will know

asher-ben-shooshan, 2020-02-06:
בס"ד
Hi
We have tried to upload the bit manipulation extension files via another user adimay162@mellanox.com<mailto:adimay162@mellanox.com>
And we have tried to sign Mellanox company on the CLA yesterday with the signed authorized of Mellanox
Company Carlos Nemerovsky Director chip design.
But today we accept the message from Google that the CLA is already signed by Mellanox by Mike Dubman VP SW Mellanox and that we
Should contact him.
We talked to him but he does not know what he should do .
Can you please guide us what we and Mike dubman should do in order to sign
The CLA and to contribute to Google the bit manipulation files.
Thanks

taoliug, 2020-02-12:
Sorry for the late response, can you check below information helpful?

**How do I add authorized contributors for my company?**
First, add their email address to the Google Group that you submitted with the corporate CLA. This is most commonly their company address.

Instruct contributors to use that same email address for any contributions to Google projects that are made on behalf of the company. For example, for projects that use git for source control, they may need to use the git config command to set their email address.

Additionally, if contributing to a project that is hosted on GitHub, contributors should add that email address to their GitHub account. This will allow us to map their GitHub username to the Corporate CLA that their contribution is covered by.

taoliug, 2020-02-12:
you can find more information here
https://cla.developers.google.com/about#add-contributors

taoliug, 2020-03-22:
this issue is now resolved. #518

closed 2020-03-22 by taoliug