chipsalliance/riscv-dv #496: sections management

agrobman, 2020-02-27:
It will make my life much easier if the test generator will stop using .pushsection/.popsection and use 
just .section statements.
the same section name can be used multiple times in the test source - assembler will combine the pieces to one continuous section 

We may need to put code to various locations, not necessary continuous to be able to test fetch unit and instruction caches..

agrobman, 2020-02-27:
regarding multiple text sections:

Our CPU has Instruction RAM and Instruction Cache, to test combined operation and ability to run code from both we need tests, where the code is placed in specific locations.
As now all test code is placed in one code section ..

taoliug, 2020-03-02:
Do you use generated data sections in your simulation?  .pushsection/.popsection was added to ensure a separate section for each data region, so that you can easily link them to match the actual memory map. 

For text sections, seems like you prefer to have multiple sections? One easy way I can do is to divide the main program and other subprograms to two separate sections.

agrobman, 2020-03-02:
I do use the data sections, but I don't like .push/popsections commands - our simulation environment doesn't understand them. We do process .sections, which we extended to setup their addresses and thus we generate linker command file automatically.

So please, use just .section commands.

The multiple test sections are needed to test icache, branch, BTB  and fetch units to create address bits aliasing.

What we really need is randomly created multiple text sections, to  make CPU jump between them in pseudo random manner as an option/or test parameter

taoliug, 2020-03-04:
I created a PR to use .section for all data sections, please add +use_push_data_section=0 to enable it. My original intention of using push/pop section is to allow linking each data section to different memory space, so that you can test load/store from different memories. This is not possible if all data pages are placed in a single data section.

agrobman, 2020-03-04:
“

I created a PR to use .section for all data sections, please add +use_push_data_section=0 to enable it. My original intention of using push/pop section is to allow linking each data section to different memory space, so that you can test load/store from different memories. This is not possible if all data pages are placed in a single data section.

“

I don’t see any correlation between pushsection/popsection usage vs .section  and “single data section” . Both .pushsection and .section create/continue a section, but the last doesn’t require .popsection statement and more common to use …

agrobman, 2020-03-05:
Hi,

Look like the new config removes data sections altogether .. I just asked to replace .pushsection/.popsection by .section

agrobman, 2020-03-10:
Hi,

I don't like the change - there is no separate data sections at all - everything is in section .data now.

As I asked before we need to execute L/S mixes targeted to different types of memory. We have normal external memory, we have I/O type external memory,  we have internal RAM. We have shared memory regions . Now we can't map data spaces to where we want.

I just asked not to use .pushsection/popsection commands and use .section instead!

Also .align 12 is too excessive for stack sections, I think. Initial SP can  be maximum aligned to DW ( 8bytes)   

can I get .sections back ?

agrobman, 2020-03-10:
Ok, I use  +use_push_data_section=1 to restore previous behavior

taoliug, 2020-03-11:
This is solved now, please check if solves the issue.

taoliug, 2020-03-11:
Please comment if anything else needs to be addressed for this issue.

agrobman, 2020-03-11:
I think we are done for now, but we will need to return to the section management later - 
1) generate random text sections and jump between them;
2) program PMP  regions and permissions randomly and create sections to exercise these permissions for code and data,  machine mode and user mode.  Permissions exercise may include jumps to not valid addresses and crossing the regions boundaries sequentially, etc ..

taoliug, 2020-03-12:
Sounds good, I am closing this issue for now. We can discuss other issues in a separate thread.

closed 2020-03-12 by taoliug