chipsalliance/riscv-dv #511: AMO tests  generate single address for all amo instructions

agrobman, 2020-03-10:
Hi,

with 1 amo section the generator generates only one address for amo instructions:

```
UVM_INFO [tracer] @451800.0ns: => @451800 #656 C0 8000137c [c59ca72f] amominu.w.aq x14, x25, (x25) [0xf0041000]	OK
UVM_INFO [tracer] @454100.0ns: => @454100 #659 C0 80001384 [658cadaf] amoand.w.aq x27, x24, (x25) [0xf0041000]	OK
UVM_INFO [tracer] @454800.0ns: => @454800 #660 C0 80001388 [644caeaf] amoand.w.aq x29, x4, (x25) [0xf0041000]	OK
UVM_INFO [tracer] @455600.0ns: => @455600 #663 C0 80001394 [e5fca22f] amomaxu.w.aq x4, x31, (x25) [0xf0041000]	OK
UVM_INFO [tracer] @456300.0ns: => @456300 #664 C0 80001398 [658ca4af] amoand.w.aq x9, x24, (x25) [0xf0041000]	OK
UVM_INFO [tracer] @457000.0ns: => @457000 #666 C0 800013a0 [042ca0af] amoadd.w.aq x1, x2, (x25) [0xf0041000]	OK

```

I was able to change this by updating 
riscv-amo_instr_lib.sv

```
  constraint addr_range_c {
    data_page_id < max_data_page_id;
    (data_page_id & 3) == 0; // AG: aligned only
    base inside {[0 : max_load_store_offset-1]};
  }

  function void pre_randomize();
    data_page = cfg.amo_region;
    max_data_page_id = data_page[0].size_in_bytes; // AG:
  endfunction

  function void post_randomize();
    gen_amo_instr();
    // rs1 cannot be modified by other instructions
    if(!(rs1_reg inside {reserved_rd})) begin
      reserved_rd = {reserved_rd, rs1_reg};
    end
    add_mixed_instr(num_mixed_instr);
    add_rs1_init_la_instr(rs1_reg, 0, data_page_id); // AG
    super.post_randomize();
  endfunction
```

PS. data_page_id variable name isn't really good now - it's offset in the amo page ...
PSS constraint addr_range_c needs enhancements :
1) it should create W/DW aligned address depending on instruction size most of the time;
2) it should create misaligned addresses probably 0.1 -1% of the time to check exception generation. 
PSSS. changes are marked by //AG comment

There is another problem that with 2 harts system probability to execute AMO instructions from different harts on the same location back to back is pretty low ...

taoliug, 2020-03-11:
Please try the fix, now it's possible to back-to-back AMO operations with the same or difference addresses. I have also increased the ratio of AMO sequences to increase the possibility of multi-harts accessing the same location.

The unaligned AMO operation will be add later as an error test

agrobman, 2020-03-11:
I'm getting following run time error:

```
UVM_INFO riscv-dv/src/riscv_asm_program_gen.sv(1382) @ 0: reporter [asm_gen] Insert directed instr stream riscv_amo_instr_stream 6/609 times
      offset[i] inside {[0 : data_page[data_page_id].size_in_bytes- 1]};
                                                  |
xmsim: *E,RNDCNSTE (riscv-dv/src/riscv_amo_instr_lib.sv,48|50): Randomization constraint has this error, which will cause the randomize function to return 0 and no new rand values will be set:
A reference to a rand variable in an array index is not currently supported in this context.
          `DV_CHECK_RANDOMIZE_FATAL(new_instr_stream)
                                                    |
xmsim: *W,SVRNDF (riscv-dv/src/riscv_asm_program_gen.sv,1394|52): The randomize method call failed. The unique id of the failed randomize call is 136.
Observed simulation time : 0 FS + 44

```

taoliug, 2020-03-11:
It should be fixed now, please let me know if it works for you.

agrobman, 2020-03-11:
It works now, thanks.

closed 2020-03-11 by taoliug