chipsalliance/riscv-dv #516: unimplemented instructions :  set bits [11:7] to 0

agrobman, 2020-03-11:
Hi,

We have some privet instructions, implemented in our processor. Sometimes unimplemented  class generates these instructions, which corrupt reserved registers. For safe operation, it will be nice to generate RD=0.

 ```
   constraint instr_bit_assignment_c {
    solve opcode before instr_bin;
    solve func3 before instr_bin;
    solve func7 before instr_bin;
    solve c_msb before instr_bin;
    solve c_op before instr_bin;
    instr_bin[11:7] == 0; //AG: do not update target it's hard to prevent bitmanip instructions to be generated
    if (compressed) {
      instr_bin[1:0] == c_op;

```

I added this constraint for that ...

I don't think this will cause a lot of coverage loss ...

taoliug, 2020-03-11:
These instructions are generated in the illegal instruction test. Since this falls into the category of custom instructions, I 'd suggest to add some constraint to the illegal instruction binary to avoid generating opcode that's used by custom instructions.

https://github.com/google/riscv-dv/blob/master/src/riscv_illegal_instr.sv#L77

Assigning RD to 0 could hide issue like illegal instruction change the RD value by mistake.

agrobman, 2020-03-11:
“

These instructions are generated in the illegal instruction test. Since this falls into the category of custom instructions, I 'd suggest to add some constraint to the illegal instruction binary to avoid generating opcode that's used by custom instructions.

“

We have too many of these, and I don’t want to deviate from master branch too much, for reasons I’ve mentioned before.

“

Assigning RD to 0 could hide issue like illegal instruction change the RD value by mistake.

“

This won’t hide any decoder problems as we run with ISS, which supposed to take illegal exception on unimplemented opcodes.

Unimplemented opcodes are defined  by other instruction bits, not by RD field, unless RISCV defines illegal instruction forms with RD=0, which I’m not aware of …

taoliug, 2020-03-11:
For illegal instruction, we want to make sure the exception is raised, and the value of RD should not be changed because the instruction should not be executed. The later issue could only be caught if RD is not zero. 

To minimize the customization effort, I can provide an array of reserved opcode in the riscv_core_setting file, you can add your reserved opcode there. What do you think?

agrobman, 2020-03-11:
For illegal instruction, we want to make sure the exception is raised, and the value of RD should not be changed because the instruction should not be executed. The later issue could only be caught if RD is not zero.

AG: what if the faulty CPU updates another GPR, when RD=0 ; these types of concerns are open-ended; That’s why we run randoms against reference model.

To minimize the customization effort, I can provide an array of reserved opcode in the riscv_core_setting file, you can add your reserved opcode there. What do you think?

Problem is that you already changed content/format and location of the core setting file at least twice.

Besides, we have variants of our cores which do and don’t implement these instructions, but we have one location of RISCV-DV code and one wrapper script to run all of them.

If you don’t want this constraint on permanent basis, we can have a plusarg/config bit that enables it …

taoliug, 2020-03-11:
Are all the custom instructions using a small set of opcode? I can provide plusargs like this
+reserved_opcode=0000000,010010,....
I still prefer to avoid hard coding rd to zero for illegal instructions. BTW, based on the instruction format, bits[11:7] may not always mean rd.

agrobman, 2020-03-11:
In our case these are bitmanip instructions which definition already changed 3 times during life of these cores …

taoliug, 2020-03-11:
Are they all bitmanip instructions? We will support it shortly, For now I can modify the illegal instruction class to avoid generating bitmanip instructions.

taoliug, 2020-04-10:
I am closing this issue now, please try to run with rv32imcb target to see if you still have this problem.

closed 2020-04-10 by taoliug