chipsalliance/riscv-dv #520: Support Bit-manipulation extension

taoliug, 2020-03-14:
File this bug to track the progress of Bit-manipulation extension support. 
Thanks for the contribution from Mellanox, we have got the basic Bit-manipulation instruction class ready (#518). Following work:
- Create a target for B-extension
- Add tests for B-extensions. 
- [Get ovpsim setup and pass ISS sim.](https://github.com/riscv/riscv-bitmanip#imperas-riscvovpsim-reference-simulation)
- Update document
- Add functional coverage model for B-extension

Please leave your comment here if you have suggestion or feature request on B-extension testing.

taoliug, 2020-03-14:
A rv32imcb target is created, OVPsim setting is also updated.

taoliug, 2020-04-10:
I believe below steps are done already.

- Create a target for B-extension
- Add tests for B-extensions.
- Get ovpsim setup and pass ISS sim.

weicaiyang, 2020-04-10:
> I believe below steps are done already.
> 
> * Create a target for B-extension
> * Add tests for B-extensions.
> * Get ovpsim setup and pass ISS sim.

Yes, these are done.

taoliug, 2020-04-10:
Add one item: Enable spike & ovpsim co-simulation flow for B-extension.
When you update the document, please provide the links for the required toolchain

taoliug, 2020-05-06:
Closing this bug as all required work is done. Thanks Weicai!

closed 2020-05-06 by taoliug