chipsalliance/riscv-dv #547: Support for Multi hart ISS Simulation

saurabhsingh-pvips, 2020-04-23:
The support for multi hart assembly generation is provided. But multi hart ISS simulation is pending.

One way I can see is to dump assembly program to different memory regions and let each thread access it's own region. This implies no shared space access.

Queries:
-  Can this assembly programs be bifurcated per hart and dumped to different locations via [this](https://htmlpreview.github.io/?https://github.com/google/riscv-dv/blob/master/docs/build/singlehtml/index.html#setup-the-memory-map)?
- For ISS simulation, can we split the assembly program for different harts and run each hart independently on ISS simulator? Will this work?

taoliug, 2020-04-23:
If you don't care about shared memory access, I think you can run the same program multiple times with ISS, each time with a different hart ID setting. The program will start with reading the hart ID register from the ISS/design and determine which instruction section to run.
Then you can aggregate all the trace logs and compare with the trace from RTL sim. This flow is not currently supported by riscv-dv.

saurabhsingh-pvips, 2020-04-24:
Can you elaborate on the hart ID setting? Is it implemented somewhere as a switch or some knob?

I had tried the following to run iss simulation sequentially per hart:
- Generate assembly program for two harts.
- Comment out instructions enabling `hart_id = 1`. Generate binary and run iss_sim. (This gives log for `hart_id = 0`.
```
_start:           
                  csrr x5, mhartid
                  li x6, 0
                  beq x5, x6, 0f
                 #li x6, 1
                 #beq x5, x6, 1f

0: j h0_start
#1: j h1_start
```
- Comment out instructions enabling `hart_id = 0`. Generate binary and run iss_sim. (This gives log for `hart_id = 1`.
```
_start:           
                  csrr x5, mhartid
                  #li x6, 0
                  #beq x5, x6, 0f
                  li x6, 1
                  beq x5, x6, 1f

#0: j h0_start
1: j h1_start
```

Observing the diff of the two logs, this seems to work.

Can you confirm this approach to be correct and if there can be some complication that I might have missed.

taoliug, 2020-04-26:
If you use ovpsim as ISS, you can try to set the mhartid to 1 for your 2nd run. I don't think you need to modify the generated program.
https://github.com/google/riscv-dv/blob/master/target/rv32imc/riscvOVPsim.ic#L14

saurabhsingh-pvips, 2020-04-26:
Thanks. This was much convenient.

closed 2020-04-26 by taoliug