chipsalliance/riscv-dv #552: cfg.sp without initialized may result in exceptions

gxflying, 2020-04-27:
I find that in the class riscv_push_stack_instr, the function gen_push_stack_instr is used to push data into user random sp cfg.sp,  but the cfg.sp is initialized nowhere !!
so the data pushing may result in exceptions  if the value contained in the cfg.sp is out of the memory range

udinator, 2020-05-01:
Hi @gxflying - thanks for the question.
During initial setup of the configuration object, we randomly assign a register as `cfg.sp` to act as the stack pointer, it is then put into a list of `reserved_regs` (https://github.com/google/riscv-dv/blob/master/src/riscv_instr_gen_config.sv#L635), which are not allowed to be touched by randomly generated instructions elsewhere in the program.
The `cfg.sp` register is then initialized in the `gen_init_section()` function of `riscv_asm_program_gen.sv` (https://github.com/google/riscv-dv/blob/master/src/riscv_asm_program_gen.sv#L416).

gxflying, 2020-05-02:
@udinator thanks for your reply, and  the answer  shows me what i want to konw. thank you!

closed 2020-05-02 by gxflying