chipsalliance/riscv-dv #556: Can QEMU be added as ISS?

saurabhsingh-pvips, 2020-05-05:
QEMU seems to outperform SPIKE by factor of 10 in terms of processing instructions per second. Is it possible to add support for QEMU?

One reservation I have come across is `However, it (QEMU) doesn’t provide an instruction-by-instruction trace as it spends much of its time executing native code` from [here](https://www.sifive.com/blog/risc-v-qemu-part-1-privileged-isa-hifive1-virtio).

Is this like dead end or there is a way one can still workaround and generate instruction trace to maintain the flow if riscv-dv?

taoliug, 2020-05-06:
If the trace log is not available, it won't be very useful for verification.

taoliug, 2020-05-09:
Closing this issue now, please let me if you want to explore more about it.

closed 2020-05-09 by taoliug