chipsalliance/riscv-dv #561: Vector Widening and Narrowing instructions when SEW == 128

Sustrak, 2020-05-07:
As the RISC-V specs says

For widening:

> The destination vector register group results are arranged as if both SEW and LMUL were at twice their current settings (i.e., the destination element width is 2*SEW, and the destination vector register group LMUL is 2*LMUL).

For narrowing

> If (2*LMUL > 8), or (2 * SEW) > ELEN, an illegal instruction exception is raised.

When setting the SEW to 128, widening instructions still are generated, same with narrowing when setting a ELEN which is less than 2*SEW

taoliug, 2020-05-29:
We are working on updating the narrowing/widening instruction support, expect to push to upstream in a few days.

taoliug, 2020-06-07:
This should be solved by https://github.com/google/riscv-dv/pull/598. Please reopen the issue if you see this problem.

closed 2020-06-07 by taoliug