chipsalliance/riscv-dv #564: mtvec handler code alignment needs to be configurable

simon5656, 2020-05-08:
In riscv_core_setting.sv I can set 
`mtvec_mode_t supported_interrupt_mode[$] = {VECTORED};`
and the generated code has an alignment of
`.align 2`

 I need different alignment.

I have worked around it with:
` -sim_opts \"+tvec_alignment=8\"`

I think there should be another line in the riscv_core_setting.sv like
`int mtvec_align = 8;`

thanks
Simon

taoliug, 2020-05-09:
Like you said, it can be changed by adding sim options. The configuration of the mtvec alignment is already provided in the riscv_instr_gen_config class.
https://github.com/google/riscv-dv/blob/master/src/riscv_instr_gen_config.sv#L69

closed 2020-05-09 by taoliug

simon5656, 2020-05-14:
I dont understand why you reject this request - I thought the riscv_core_setting.sv was for setting config options for the core - from the list of options in the ISA specs - for example supported ISA, interrupt mode support (DIRECT, VECTORED) etc.

The idea of the riscv_core_setting.sv is to config the generator/coverage for options in the core.

The ISA specs says for VECTORED interrupts the core can select the alignment - so why dont you want to have this config option in the riscv_core_setting.sv file?

Why make it a simulator option - that is completely wrong - it is nothing to do with the simulator but the core. If the core option on vectored interrupt alignment is not in the core config - then there is no real point of having the riscv_core_setting.sv file and taregt directory at all - and all config options should be done with +args... (which is not the correct outcome).

So please - reconsider my suggestion - the plan surely is that the riscv_core_setting.sv is the place for config options - and in future it will be replaced with device.yaml.

thanks for your understanding.

Simon.