chipsalliance/riscv-dv #572: Move `sext.b/h` bitmanip instructions to Zbb sub group

vogelpi, 2020-05-13:
These two instructions are not listed in any sub group in the version 0.92 draft of the bitmanip extension. This is probably an oversight and in the [current working draft version](https://github.com/riscv/riscv-bitmanip/blob/master/bitmanip-draft.pdf), they are part of the Zbb sub group.

Currently, they are listed in the Zbe sub group inside RISCV-DV
https://github.com/google/riscv-dv/blob/b7e68eff1997bd9bea54daabdf0316574a33d2ca/src/isa/riscv_b_instr.sv#L479-L480

Other than the name (sext and bext) there is not really a similarily between these instructions. The former is a **s**ign **ext**end, the latter a **b**it **ext**ract. Would it be possible to move these into Zbb?

vogelpi, 2020-05-13:
This maybe of interest for @weicaiyang and @udinator .

weicaiyang, 2020-05-13:
@vogelpi 
I was not sure how to categorize these 2 instructions. Thanks for the info.
Looks like in the working draft spec, slo[i], sro[i] have been moved to zbp. Should we also update accordingly, or it's fine to have them in zbb?

vogelpi, 2020-05-14:
Hi @weicaiyang ,
indeed, also other instructions are moved around. In this case, it is maybe best not to move them to Zbb (and also not move slo/sro). Otherwise, we are aligned to something that probably still keeps changing. The current working draft does not have a version number, so there is not really something we can refer to. And we might need to change it again when something in the working draft changes.

What about just moving sext.b/h out of Zbe into a new group called Zb_tmp or similar to indicate this is just temporary? And then we can say that we are still aligned with version 0.92 draft.

weicaiyang, 2020-05-14:
> Hi @weicaiyang ,
> indeed, also other instructions are moved around. In this case, it is maybe best not to move them to Zbb (and also not move slo/sro). Otherwise, we are aligned to something that [...]

Make sense. Will create a zb_tmp for sext.b/h. Thanks

vogelpi, 2020-05-14:
That would be great, thanks @weicaiyang !

weicaiyang, 2020-05-15:
I think we can close this now

closed 2020-05-15 by weicaiyang