chipsalliance/riscv-dv #574: mtvec handler code alignment needs to be configurable (#564)

simon5656, 2020-05-14:
[I could not see how to re-open an issue so am adding this comment here as well for issue #564]

I dont understand why you reject this request - I thought the riscv_core_setting.sv was for setting config options for the core - from the list of options in the ISA specs - for example supported ISA, interrupt mode support (DIRECT, VECTORED) etc.

The idea of the riscv_core_setting.sv is to config the generator/coverage for options in the core.

The ISA specs says for VECTORED interrupts the core can select the alignment - so why dont you want to have this config option in the riscv_core_setting.sv file?

Why make it a simulator option - that is completely wrong - it is nothing to do with the simulator but the core. If the core option on vectored interrupt alignment is not in the core config - then there is no real point of having the riscv_core_setting.sv file and taregt directory at all - and all config options should be done with +args... (which is not the correct outcome).

So please - reconsider my suggestion - the plan surely is that the riscv_core_setting.sv is the place for config options - and in future it will be replaced with device.yaml.

thanks for your understanding.

Simon.

taoliug, 2020-05-18:
I am ok with either ways. The reason I didn't put in to core_setting is because it requires changing core_setting files on the user side. I may need to refactor the core_setting file to avoid such small change impacting user code.

taoliug, 2020-07-04:
Simon, I updated the constraint to setup the alignment for vectored interrupt mode. PTAL
For now I still keep it in riscv_instr_gen_config as the core may support both DIRECT and VECTORED interrupt mode, keep it a random setting could help to test all allowed mtvec alignment cases.