chipsalliance/riscv-dv #575: issue of pointer arith and branches with RISCV-DV generated code

simon5656, 2020-05-14:
Hi Tao
We have started to move to using the latest commits of riscv-dv and have started putting some issues on the github.

This issue is a little more detailed than most -with attachments.

We have discovered a strange behavior with one of the generated testcases for rand_jump tests
We have a generated assembler file riscv_rand_test_9.S

This is assembled and linked using two different linker scripts to match the target execution
environments, the two linker scripts are supplied
link.1.ld // seems to work correctly
link.2.ld // seems to generate incorrect executable

We have tried to reverse engineer the intent, and it would appear that there is a pointer
arithmetic expression to determine whether a branch is taken or not.
There could be 2 possible issues with this, firstly the calculated address from an AUIPC
has overflowed, or the branch condition is using a signed expression

The calculation of one of the branch terms looks like this
Info 229: 'riscvOVPsim/cpu', 0x000000008000a3b8(sub_7_stack_p+8c): Machine 95875a97 auipc   s5,0x95875
Info   s5 f673a4a2 -> 1587f3b8
This looks like the address has overflowed ?

In the alternate case
Info 229: 'root/cpu0', 0x00000000000123b8(sub_7_stack_p+8c): 95875a97 auipc   s5,0x95875
Info   s5 f673a4a2 -> 958873b8
Which has not overflowed

This is then used in a branch expression
Info 431: 'riscvOVPsim/cpu', 0x000000008000a69e(sub_7_stack_p+372): Machine 0351cf63 blt     gp,s5,8000a6dc
and
Info 431: 'root/cpu0', 0x000000000001269e(sub_7_stack_p+372): 0351cf63 blt     gp,s5,126dc
respectively

Each of those branches behave differently due to the calculation in s5

I have also attached the logs

Attached
riscv_rand_test_9.S
link.1.ld
link.2.ld
ovpsim_riscv_rand_jump_test.9.log
rtl_riscv_rand_jump_test_9.S.log

thanks
Simon

[the commit we are using is c37c5f3 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #554 from udinator/fix_pmp_max_offset]

[link.1.ld.txt](https://github.com/google/riscv-dv/files/4630838/link.1.ld.txt)
[link.2.ld.txt](https://github.com/google/riscv-dv/files/4630839/link.2.ld.txt)
[ovpsim_riscv_rand_jump_test.9.log.txt](https://github.com/google/riscv-dv/files/4630840/ovpsim_riscv_rand_jump_test.9.log.txt)
[riscv_rand_jump_test_9.S.txt](https://github.com/google/riscv-dv/files/4630841/riscv_rand_jump_test_9.S.txt)
[rtl_riscv_rand_jump_test_9.S.log.txt](https://github.com/google/riscv-dv/files/4630842/rtl_riscv_rand_jump_test_9.S.log.txt)

taoliug, 2020-05-18:
The auipc instruction you pointed is just a random instruction, there's not special intention behind it, so it doesn't matter if the result overflow or not. I don't quite get the simulation setup here. Do you try to use two link scripts to link the program to different PC and run co-simulation with these two executables? In this case, the result will be different because the PC used in calculation is different.

eroom1966, 2020-05-18:
Hi @taoliug 
The problem here is that there is a branch in the code which behaves differently, depending upon where the program is linked to run from. 

The branch taken/not-taken (Dependant upon the result of auipc) is different in each case. In one if the scenarios the execution locks up in an infinite loop

Thx
Lee

taoliug, 2020-05-18:
The branch instruction based on s5 is a forward branch, which should not create an infinite loop.  rtl_riscv_rand_jump_test_9.S.log.txt does look like a timeout test, but I cannot tell whether it's because of the aupic instruction. Can you provide more information? For example, set a shorter timeout threshold and upload the full log tar file.

Imperas, 2020-05-18:
@taoliug 
I will speak with Simon regarding a shorter test, he is the expert!

The problem as I observed is that the auipc result placed into s5 is treated as (signed) positive or negative in the branch instruction
The address at which the program is linked will cause a sign difference in s5.

in one case the number overflows from negative back to positive 
Info s5 f673a4a2 -> 1587f3b8 // positive number

in the other case the number remains negative
Info s5 f673a4a2 -> 958873b8 // negative number

This means the branch (blt) instruction using s5 will differ as it is a signed comparison

This means that depending upon the target linker script which is used the result differs
using your linker script (memory located at 0x80000000), the program behaves as expected, using the linker script for the cv32e40p (memory located at 0x00000000) it does not.

Thx
Lee

taoliug, 2020-05-20:
I think different result is fine as the link script is different, it should not directly lead to infinite loop issue.  The root cause could be something else. I will need the full log debug further

taoliug, 2020-06-03:
Simon, can you sync and rerun to see if the issue is still there. We recently found a jump instruction issue in our regression, and fixed in https://github.com/google/riscv-dv/pull/590

taoliug, 2020-06-17:
Closing this issue for inactivity, free feel to reopen it if you still see this problem

closed 2020-06-17 by taoliug