chipsalliance/riscv-dv #576: Interrupts Tests not working with SweRV Core

haroonshafique-lm, 2020-05-18:
I have generated interrupts tests using "riscv_full_interrupt_test" test from base_testlist.yaml. I am using rv32imc as target. However, I am unable to run them successfully on SweRV core. On simulating the test, it stucks in an infinite loop. It would be helpful if you answer the following two questions.
1. Which configuartions need to be modified and added so that generated tests can run with any generic core?
2. How to configure the memory model (i.e. memory addresses etc) according to the memory model of the core?

mcockrell-google, 2020-05-28:
Information on configuration and memory map can be found [here](https://htmlpreview.github.io/?https://github.com/google/riscv-dv/blob/master/docs/build/singlehtml/index.html#setup-the-memory-map)

Also - we have a bi-weekly workgroup to discuss the tool, please join us tomorrow. Meeting link [here](https://meet.google.com/soh-ovxy-vez)

taoliug, 2020-05-28:
Do you have your own interrupt handling routine which is not generated by riscv-dv?

haroonshafique-lm, 2020-05-29:
> Do you have your own interrupt handling routine which is not generated by riscv-dv?

I do not have my own interrupt handling routine. Does google riscv-dv not generate this on its own?

haroonshafique-lm, 2020-05-29:
> Information on configuration and memory map can be found [here](https://htmlpreview.github.io/?https://github.com/google/riscv-dv/blob/master/docs/build/singlehtml/index.html#setup-the-memory-map) [...]

Thanks for the memory map information. 
What is the time and corresponding time zone for meeting?

mcockrell-google, 2020-05-29:
9am PST

- Matt

taoliug, 2020-05-29:
Sorry, I thought you are from SweRV team, they are using their own interrupt handling routine instead the one generated by riscv-dv. Can you elaborate a bit more about the infinite loop problem? 

You can configure the memory map here:
https://github.com/google/riscv-dv/blob/master/src/riscv_instr_gen_config.sv#L112

haroonshafique-lm, 2020-05-29:
> 9am PST - Matt
> […](#)

Thanks.

haroonshafique-lm, 2020-05-29:
> Sorry, I thought you are from SweRV team, they are using their own interrupt handling routine instead the one generated by riscv-dv. Can you elaborate a bit more about the infinite loop problem? [...]

By infinite loop problem I mean that the stimulation never stops. I monitored the mcycle, mcause and mie CSR. Following are the values corresponding to last test generated from google rv-dv:
- mcycle keeps on incrementing by 1 
- mcause has a value of 11 (in decimal format)
- bit 3 (software interrupt), 7 (timer interrupt) and 11 (External interrupt) of mie all are 1 at the same time which is odd. 

And mcause = 11 corresponds to environment call from Machine mode.

I could not understand which things need to be configured for running interrupts test on SweRV core. It would be very much helpful, if you tell me which configurations need to be modified and added so that generated tests can run with any core.

taoliug, 2020-06-02:
By default the interrupt handler will just read some privileged CSR(mstatus, mcause etc) and resume execution. mcause = 11 indicates the program has reached the end point. TB could terminate the test when mcause = 11 is detected. How do you terminate your RTL simulation for normal test?

agrobman, 2020-06-04:
hi,

You need :
1) configure PIC - interrupt controller in SweRV core; (See Swerv PRM how to do this)
2) drive interrupt requests someway; 
3) write interrupt handler such that the interrupt source dis-asserts interrupts before the CPU exits interrupt handler and place the handler at external interrupt vector address, enable interrupts. in mstatus register

taoliug, 2020-06-12:
Thanks for Alex for the reply. Closing this issue for now, please reopen it if you have any more questions.

closed 2020-06-12 by taoliug