chipsalliance/riscv-dv #582: UVM_FATAL error occurs when VSETVL instruction has been randomized.

dariuszst, 2020-05-28:
When the VSETVL (probably also VSETVLI) instruction has been randomized the following error is generated:

`> UVM_FATAL $$STRING$$/src/riscv-dv/src/isa/riscv_instr.sv(418) @ 0: reporter [riscv_VSETVL_instr] Unsupported format VSET_FORMAT [VSETVL]`

This occures for the rv64gcv target, probably because of the specific  format (VSET_FORMAT) of this instruction defined in the file riscv-dv/src/isa/rv32v_instr.sv (17):
```
// Vector CSR access instruction
`DEFINE_INSTR(VSETVLI,    VSET_FORMAT, CSR, RV32V)
`DEFINE_INSTR(VSETVL,     VSET_FORMAT, CSR, RV32V)
```
Theese instructions are registered as a regular riscv_instr class objects  - file riscv-dv/src/riscv_defines.svh (67): 
```
// Regular integer instruction
`define DEFINE_INSTR(instr_n, instr_format, instr_category, instr_group, imm_tp = IMM)  \
  class riscv_``instr_n``_instr extends riscv_instr;  \
    `INSTR_BODY(instr_n, instr_format, instr_category, instr_group, imm_tp)
```
Then, during instruction stream writing to the assembler program, the "convert2asm" function  (file riscv-dv/src/isa/riscv_instr.sv(349)) is used that contains "case" expression:
```
if(category != SYSTEM) begin
      case(format)
        J_FORMAT, U_FORMAT : // instr rd,imm
          asm_str = $sformatf("%0s%0s, %0s", asm_str, rd.name(), get_imm());
        I_FORMAT: // instr rd,rs1,imm
          if(instr_name == NOP)
            asm_str = "nop";
          else if(instr_name == WFI)
            asm_str = "wfi";
          else if(instr_name == FENCE)
            asm_str = $sformatf("fence"); // TODO: Support all fence combinations
          else if(instr_name == FENCE_I)
            asm_str = "fence.i";
          else if(category == LOAD) // Use psuedo instruction format
            asm_str = $sformatf("%0s%0s, %0s(%0s)", asm_str, rd.name(), get_imm(), rs1.name());
          else if(category == CSR)
            asm_str = $sformatf("%0s%0s, 0x%0x, %0s", asm_str, rd.name(), csr, get_imm());
          else
            asm_str = $sformatf("%0s%0s, %0s, %0s", asm_str, rd.name(), rs1.name(), get_imm());
        S_FORMAT, B_FORMAT: // instr rs1,rs2,imm
          if(category == STORE) // Use psuedo instruction format
            asm_str = $sformatf("%0s%0s, %0s(%0s)", asm_str, rs2.name(), get_imm(), rs1.name());
          else
            asm_str = $sformatf("%0s%0s, %0s, %0s", asm_str, rs1.name(), rs2.name(), get_imm());
        R_FORMAT: // instr rd,rs1,rs2
          if(category == CSR) begin
            asm_str = $sformatf("%0s%0s, 0x%0x, %0s", asm_str, rd.name(), csr, rs1.name());
          end else if(instr_name == SFENCE_VMA) begin
            asm_str = "sfence.vma x0, x0"; // TODO: Support all possible sfence
          end else begin
            asm_str = $sformatf("%0s%0s, %0s, %0s", asm_str, rd.name(), rs1.name(), rs2.name());
          end
        default: `uvm_fatal(`gfn, $sformatf("Unsupported format %0s [%0s]",
                                            format.name(), instr_name.name()))
      endcase
```
The "VSETVL" instruction is a CSR category instruction with "VSET_FORMAT" format that means that a "default:" clause of a case statement will be used and fatal error thrown.

The "VSETVL" is randomized during the randomization of one of the subprograms - it seems that there is no constraint disabling that instruction what means  "VSETVL" is allowed in this context.
If yes, then to get rid of the UVM_FATAL error the format of "VSETVL" instruction should be changed in the file riscv-dv/src/isa/rv32v_instr.sv (17) or in a convert2asm function (riscv-dv/src/isa/riscv_instr.sv(349)) the extra clause should be added to the case statement.

Probably the case statement in the convert2bin function ( riscv-dv/src/isa/riscv_instr.sv(549)) should be modified also. That functions by the way semms to be wrongly commented as:
`// Convert the instruction to assembly code`

taoliug, 2020-05-29:
Will temporarily disable this instruction in random instruction generation. We can add it back later to randomly changing vector CSR setting.

taoliug, 2020-06-03:
BTW, by default this instruction should be disabled, do you have your own testlist?
https://github.com/google/riscv-dv/blob/master/target/rv64gcv/testlist.yaml#L46

If yes, please set no_csr_instr to 1 in your testlist. We will add support to this instruction soon.

dariuszst, 2020-06-03:
No, I don't have own testlist. I've started default test set for rv64gcv target with a command:
 >run.py --simulator riviera --steps gen --target=rv64gcv
As I know the no_csr_instr=1 option is set there already.

taoliug, 2020-06-03:
I see the problem, please check if it's fixed. BTW, there are some other issue related to rv64gcv (#588 )

PR #591 (2020-06-03): Fix VSETVL generation issue
fix #582
closed 2020-06-03 by taoliug