chipsalliance/riscv-dv #588: Uninitialized registers in riscv_rand_jump_test (and others)

dpetrisko, 2020-06-02:
Hi, 

When I run the generator for riscv_rand_jump_test, the attached program is generated.  The problem is that 'init_supervisor' is called before 'init'/'init_gpr'.  In particular: 

```
000000008001e9e4 <fix_pte_done>:
    8001e9e4:   03ef1f13            slli    t5,t5,0x3e
```
is called before init_gpr, meaning t5 is uninitialized.  This results in an undefined value in t5, corrupting the rest of the test.  When run with spike/dromajo, the value in t5 is assumed to be 0 to start, so this issue does not occur.

[riscv_rand_jump_bug.zip](https://github.com/google/riscv-dv/files/4719191/riscv_rand_jump_bug.zip)

taoliug, 2020-06-03:
I can reproduce this issue, it seems to be something related to page table setup, will debug further

taoliug, 2020-06-03:
I have submitted the fix, please check if it works for you. BTW, I have some issue with simulating with ovpsim, but spike works fine.

dpetrisko, 2020-06-03:
Yes, that fixed my issue.  Thanks!

closed 2020-06-03 by dpetrisko