chipsalliance/riscv-dv #592: Questasim doesn't recognise the assembler code

ShoaibSaifu, 2020-06-03:
The logfile contains the following messages:
# /tmp/xxx@host_dpi_4212/linux_x86_64_gcc-9.2.0/export_tramp.S: Assembler messages:
# /tmp/xxx@host_dpi_4212/linux_x86_64_gcc-9.2.0/export_tramp.S:18: Error: unrecognized opcode `movq svdpi_curr_tramp@GOTPCREL(%rip),%rax'
# /tmp/xxx@host_dpi_4212/linux_x86_64_gcc-9.2.0/export_tramp.S:19: Error: unrecognized opcode `movq (%rax),%rax'
# /tmp/xxx@host_dpi_4212/linux_x86_64_gcc-9.2.0/export_tramp.S:26: Error: unrecognized opcode `movq $53000992,%rax'
# /tmp/xxx@host_dpi_4212/linux_x86_64_gcc-9.2.0/export_tramp.S:27: Error: unrecognized opcode `pushq %rax'
# /tmp/xxx@host_dpi_4212/linux_x86_64_gcc-9.2.0/export_tramp.S:28: Error: unrecognized opcode `movq svdpi_curr_tramp@GOTPCREL(%rip),%rax'
# /tmp/xxx@host_dpi_4212/linux_x86_64_gcc-9.2.0/export_tramp.S:29: Error: unrecognized opcode `popq (%rax)'
# /tmp/xxx@host_dpi_4212/linux_x86_64_gcc-9.2.0/export_tramp.S:30: Error: unrecognized opcode `movq $53000992,%rax'
# /tmp/xxx@host_dpi_4212/linux_x86_64_gcc-9.2.0/export_tramp.S:31: Error: unrecognized opcode `jmp *(%rax)'

No luck with with riscv64 toolchain either 

DpiCppInstall is pointing to the expected riscv32-unknown-linux-gnu-gcc

taoliug, 2020-06-03:
I don't know how this related to QuestaSim or riscv-dv, can you elaborate a bit more?

agrobman, 2020-06-04:
I guess you are compiling code with not RISCV tool chain

ShoaibSaifu, 2020-06-04:
@taoliug , i am using riscv-dv to generate asm tests using the command below 
python3 run.py --simulator questa --test riscv_rand_instr_test --iss ovpsim --isa rv32imc --mabi ilp32

the same toolchain works with Cadence irun but not with Questa. 

@agrobman i made sure that gcc is pointing to the riscv toolchain .

-DpiCppInstall (ModelSim.ini) is pointing to the expected riscv32-unknown-linux-gnu-gcc

as you look into the log : 
/tmp/xxx@host_dpi_4212/linux_x86_64_gcc-9.2.0/export_tramp.S

the ASM file is created in the /tmp : is there a way to change that ? with my sims using cadence irun ASM file and the object file are both under instr_gen and not /tmp

agrobman, 2020-06-04:
I don’t think Questa should use riscv tools. RISCV tools should compile only RISCV-DV output. DPI routines need to be compiled by server native GCC compiler

taoliug, 2020-06-07:
+1 to Alex's comment, please use native gcc compiler for Questasim.

closed 2020-06-07 by taoliug