chipsalliance/riscv-dv #595: When choosing to support sv32, spike simulation timeout

fenghao8023, 2020-06-04:
The configuration of my run riscv-dv is as follows:
// Processor feature configuration
//-----------------------------------------------------------------------------
// XLEN
parameter int XLEN = 32;

// Parameter for SATP mode, set to BARE if address translation is not supported
parameter satp_mode_t SATP_MODE = SV32;

// Supported Privileged mode
privileged_mode_t supported_privileged_mode[] = {USER_MODE, SUPERVISOR_MODE, MACHINE_MODE};

// Unsupported instructions
riscv_instr_name_t unsupported_instr[];

// ISA supported by the processor
riscv_instr_group_t supported_isa[$] = {RV32I, RV32M, RV32C, RV32A, RV32F, RV32D};

// Interrupt mode support
mtvec_mode_t supported_interrupt_mode[$] = {DIRECT, VECTORED};

// The number of interrupt vectors to be generated, only used if VECTORED interrupt mode is
// supported
int max_interrupt_vector_num = 16;

// Physical memory protection support
bit support_pmp = 0;

// Debug mode support
bit support_debug_mode = 0;

// Support delegate trap to user mode
bit support_umode_trap = 0;

// Support sfence.vma instruction
bit support_sfence = 1;

// Support unaligned load/store
bit support_unaligned_load_store = 0;

When I run it, the error message is as follows:
`Thu, 04 Jun 2020 15:02:42 ERROR    Timeout[10s]: /home/xiexz/project/riscv-isa-sim/install/bin/spike --log-commits --isa=rv32imafdc -m0x40000000:0x50000000 -l out_2020-06-04/asm_tests/riscv_rand_instr_test_0.o &> out_2020-06-04/spike_sim/riscv_rand_instr_test.0.log`

I checked riscv_rand_instr_test.0.log and found that spike encountered an illegal instruction exception:
`core   0: 0x00000000400137d6 (0x30200073) mret`
`3 0x400137d6 (0x30200073)`
`core   0: 0x0000000000006222 (0x00000000) c.addi4spn s0, sp, 0`
`core   0: exception trap_illegal_instruction, epc 0x0000000000006222`
`core   0:           tval 0x0000000000000000`
`core   0: 0x0000000040013000 (0x34051573) csrrw   a0, mscratch, a0`
`3 0x40013000 (0x34051573) x10 0x4002de7c`

I found that the problem seems to occur after the execution of the mret instruction, spike enters smode and start SV32 virtual address access failed. How can this problem be solved? Is there a problem with my configuration?

taoliug, 2020-06-12:
Looks like there's some issue with SV32 page table generation, will debug further and provide a fix

fenghao8023, 2020-06-12:
> Looks like there's some issue with SV32 page table generation, will debug further and provide a fix

closed 2020-06-12 by fenghao8023

betak, 2021-09-30:
This issue was closed, but was it ever resolved?  I'm seeing the same issue and the problem seems to be that PTE directory entries have the A, D, and U bits set which the privilege spec says should be cleared by software.  If I comment out this check in Spike, the page tables are successfully walked and the test completes.

I see that there's a constraint in riscv_page_table_entry.sv, sw_legal_c, that should keep these bits clear when the read/write/execute bits are clear (directory pte), but it doesn't seem to be having an effect.  If it does, something is happening to the entries during post randomization.  I'm no expert in verilog, so I'm not sure how to debug this and root cause what is happening.