chipsalliance/riscv-dv #597: Customizing the travis.yml configuration for pygen

aneels3, 2020-06-07:
Travis CI build is not enabled for upstream repo as of now but in the future, it will break for pygen related PRs.
Travis CI is breaking because of  `- sphinx-build -E -W -b linkcheck docs/source build` .
The build logs are here <https://travis-ci.org/github/aneels3/riscv-dv/jobs/695695999>
I have commented out the `- sphinx-build -E -W -b linkcheck docs/source build`  of script section of tracis.yml file and its working fine.
<https://travis-ci.org/github/aneels3/riscv-dv/jobs/695700126>

Is it possible to customize the CI configuration for pygen related PRs only without eliminating the sphinx-build?
@danghai @taoliug

danghai, 2020-06-07:
The sphinx-build is wrong because it has a broken link.
```
/home/travis/build/aneels3/riscv-dv/docs/source/end_to_end_simulation.rst.rst:30:broken link: https://github.com/lowRISC/ibex/blob/master/dv/uvm/Makefile#L68 (Anchor 'L68' not found)
```
I think it is good to check the sphinx build. We just fix the link. It will pass the test

aneels3, 2020-06-07:
Got it!
Thanks @danghai

aneels3, 2020-06-07:
Still, I am getting the same error
<https://travis-ci.org/github/aneels3/riscv-dv/jobs/695762073>

> The sphinx-build is wrong because it has a broken link.
> https://github.com/lowRISC/ibex/blob/master/dv/uvm/Makefile#L68 at `end_to_end_simulation.rst.rst` line 30
> 
> ```
> /home/travis/build/aneels3/riscv-dv/docs/source/end_to_end_simulation.rst.rst:30:broken link: https://github.com/lowRISC/ibex/blob/master/dv/uvm/Makefile#L68 (Anchor 'L68' not found)
> ```
> 
> I think it is good to check the sphinx build. We just fix the link. It will pass the test

danghai, 2020-06-07:
I think @udinator need to fix that link because he documented the `end_to_end_simulation.rst.rst`.
I see the it passed the `flake8`. 
```
Warning, treated as error:
/home/travis/build/aneels3/riscv-dv/docs/source/end_to_end_simulation.rst.rst:30:broken link: https://github.com/lowRISC/ibex/blob/master/dv/uvm/Makefile#L68 (Anchor 'L68' not found)
The command "sphinx-build -E -W -b linkcheck docs/source build" exited with 2.
0.34s$ pip uninstall -y riscv-dv
Uninstalling riscv-dv-1.0:
  Successfully uninstalled riscv-dv-1.0
The command "pip uninstall -y riscv-dv" exited with 0.
0.57s$ flake8 --show-source pygen/pygen_src/ --config=.flake8
The command "flake8 --show-source pygen/pygen_src/ --config=.flake8" exited with 0.
```
Do you modify the source? If yes, you can create PR. Just ignore the `sphinx-build`.

aneels3, 2020-06-08:
I have modified the pygen source files for flake8 related issues only.
Should I create the PR?

taoliug, 2020-06-08:
yes, please go ahead to create the PR, thanks

taoliug, 2020-10-14:
Any update on this issue?

aneels3, 2020-11-02:
Hi @taoliug and @udinator 
I think the issue is fixed in #738 .
If I am correct then you can close this issue for now.