chipsalliance/riscv-dv #603: Initialization of vector registers don't scale well

Sustrak, 2020-06-10:
With the new implementation of memory operations for the vector extension the initialization of the vector registers has changed.

Before the update, vector registers were initialized using the instruction `vmv.v.x` and so all the elements in the vector register had the same value. With the new update each element is generated randomly using the instructions `vmv.v.x` and `vslide1up`.

One problem with this is when generating code for a vector machine with a big VLEN, for example in our case VLEN = 16384 (256x64) the code for initilizing the vector registers is +20k LOC long.

It would be nice to have different configurations for initilizing the vector registers I think of 3 of them:

1. Using only `vmv.v.x`.
2. Using `vmv.v.x` and `vslide1up`.
3. Using `vle.v`.

For vle I implemented this in the `init_vec_gpr` function:
```
for (int v = 1; v < NUM_VEC_GPR; v++) begin
      int region = $urandom_range(0, 4);
      instr_stream.push_back($sformatf("%0sla t0, region_%0d", indent, region));
      instr_stream.push_back($sformatf("%0svle.v v%0d, (t0)", indent, v));
end
```

taoliug, 2020-06-12:
SGTM, I will add support for these three modes

taoliug, 2020-07-03:
Fixed https://github.com/google/riscv-dv/pull/627

closed 2020-07-03 by taoliug