chipsalliance/riscv-dv #618: Randomization issue in riscv_instr.py file

aneels3, 2020-06-21:
The inline randomization used in **get_rand_instr** function of **riscv_instr.sv** file. Previously We had marked it as TODO in riscv_instr.py file here 
[image]
 I have tried it in a different way but it doesn't work well for all possible cases
[image]

Need some suggestion on this.

udinator, 2020-06-22:
Hi @aneels3 thanks for bringing this up.

@hodjat91 is actually doing some investigation into a constraint-based randomization method using PyVSC library that we can use, as part of his summer internship project.
He will have some results to present on the Thursday meeting.

aneels3, 2020-06-24:
Hi, @udinator thanks for the reply.
We have also done some research on constraints using the PyVSC library. We will discuss the same in the meeting.

aneels3, 2020-06-29:
Hi @udinator 
As there were some limitation with PyVSC library for inline randomization,
I have tried to optimize the logic for this issue.

```python
        if(len(disallowed_instr) == 0):
            if(len(include_instr) > 0):
                idx = random.randrange(0, len(include_instr) - 1)
                name = include_instr[idx]
            elif(len(allowed_instr) > 0):
                idx = random.randrange(0, len(allowed_instr) - 1)
                name = allowed_instr[idx]
            else:
                idx = random.randrange(0, len(self.instr_names) - 1)
                name = self.instr_names[idx]
        else:
            try:
                name = random.choice(self.instr_names)
                if(len(include_instr) > 0):
                    name = random.choice(include_instr)
                if(len(allowed_instr) > 0):
                    name = random.choice(allowed_instr)
            except Exception:
                logging.critical("[%s] Cannot generate random instruction", riscv_instr.__name__)
                sys.exit(1)
        instr_h = copy(self.instr_template[name])
        return instr_h
```

and It's working fine.
Can we continue with this approach?

udinator, 2020-06-29:
> ```python
>         if(len(disallowed_instr) == 0):
Should we also have a `try...except` block here as well?
> if(len(include_instr) > 0):
>                 idx = random.randrange(0, len(include_instr) - 1)
>                 name = include_instr[idx]
>             elif(len(allowed_instr) > 0):
>                 idx [...]

Sure, if this method is working, then it's fine to use it as a temporary workaround for the lack of inline support in PyVSC.
Can you please add a TODO comment above this block to switch over to PyVSC once support is added?

hodjat91, 2020-06-30:
Hi @aneels3 

> As there were some limitation with PyVSC library for inline randomization

Is this regarding the randomization on some specific data-types (like lists)?

I believe it's okay to move forward using this method as it works functionally, but I think it would look better if we can eventually frame it a more SV-like syntax using PyVSC.

aneels3, 2020-07-02:
Hi @hodjat91 
In-line randomization is supported in PyVSC for a list with integer values only.
It doesn't support other data types like string, enum, etc.

I am moving with the above-mentioned approach for the time being. We will definitely frame it with PyVSC once they add the support for this issue.

taoliug, 2020-09-07:
Is there anything else needs to be done for this issue?

hodjat91, 2020-09-07:
Hi @taoliug 
I believe this issue is actually resolved and we can close it. @aneels3 would you please confirm?

aneels3, 2020-09-09:
Yes, it is being resolved with the temporary workaround. Haven't face any issue with the workaround.
I am closing this for now. I will reopen this if I find any bug in the future.

closed 2020-09-09 by aneels3