chipsalliance/riscv-dv #621: Random seed in riscv_asm_program_gen.py

pdonahue-ventana, 2020-06-23:
While the SystemVerilog code is obviously controlled by the random seed mechanism of the simulator, the experimental riscv_asm_program_gen.py program doesn't have a way for the user to specify a seed.  This is useful to be able to reproduce the same test.S file based on just a seed rather than having to preserve files on a failure and then potentially copying them around to reproduce the problem.

Thanks.

udinator, 2020-06-23:
Hi @pdonahue-ventana - yes you are correct, the code in `pygen/experimental` does not support this. This directory is just a basic proof of concept, and has completely stopped being developed/supported since mid-2019.

We are rebuilding the Python generator from the ground up in `pygen/pygen` directory to match the existing featureset of the SystemVerilog generator. This work is still in very early stages, but we will ensure to allow the user to specify a seed when we develop our randomization mechanisms.

pdonahue-ventana, 2020-06-24:
OK.  I think that this answers my question in the other issue about what's experimental and what's not.  Thanks for considering this feature request.

closed 2020-07-14 by udinator