chipsalliance/riscv-dv #622: pygen requires bitstring package

pdonahue-ventana, 2020-06-23:
https://github.com/google/riscv-dv/blob/master/pygen/experimental/README.md#install-required-packages should mention that `bitstring` is also a required package.  Thanks.

udinator, 2020-06-23:
Hi @pdonahue-ventana you are right, thanks for pointing this out - I will open a PR to add this.

Please note that `pygen/experimental` is an experimental proof of concept, and is no longer supported.

pdonahue-ventana, 2020-06-23:
OK.  I was going to play with this because it doesn't require simulator licenses and was hoping it would move from experimental to non-experimental.  Is pygen/pygen_src the path forward?  Or is that part of the same dead-end code?  (I haven't gotten far enough into this to have figured out the code structure, etc.)

Thanks.

udinator, 2020-06-24:
Got it.
`pygen/pygen_src` is currently under development, and will eventually reach feature parity with the SystemVerilog generator.

closed 2020-07-04 by taoliug