chipsalliance/riscv-dv #634: GPR results not found

jiangxuanfeng, 2020-07-06:
command: python3 run.py --iss spike
When this command is executed, there are no GPR results in the log and CSV files.Why?

taoliug, 2020-07-06:
Can you copy the screen log here? most likely some early steps went wrong.

taoliug, 2020-07-06:
BTW, please do sync to the latest and try, there was some bugs introduced a couple of days ago.

jiangxuanfeng, 2020-07-06:
There are no errors in the operation, but there are only two lines in the VCs file,
pc,instr,gpr,csr,binary,mode,instr_str,operand,pad
ffffffff80000158,,,,00000073,,ecall,,

udinator, 2020-08-06:
Hi @jiangxuanfeng can you sync to top of tree and see if this issue occurs again?

udinator, 2020-08-31:
@jiangxuanfeng I will close this issue as it has almost been a month since the last comment in the thread.
Please feel free to open another issue (or reopen this one) if this issue still comes up.

closed 2020-08-31 by udinator