chipsalliance/riscv-dv #636: PMP support for rv64gc in SV39 mode appears broken (for single & dual harts)

vandanaprabhu, 2020-07-07:
The test generation flow appears broken when I try to customize the default target rv64gc with PMP support for SV39 mode  on Mentor or Cadence simulators? 

I've pulled in upstreams upto  `d74484b - (9 days ago) Support a vetor instruction only mode (#626) - taoliug`

I see randomization failures  for assembly test generation on tests using `gen_test=riscv_rand_instr_test`  on both Questa and Xcelium in different ways
I initially tried a custom target that uses a `riscv_core_setting.sv` template derived  from target/rv64gc with dual harts  but I  see the same failures as the  default settting (num_harts=1) and testlist with `--target=rv64gc` , SV39 with PMP support. 
Notably, `riscv_arithmetic_basic_test` passes (as does `riscv_floating_point_arithmetic_test`) which doesn't use loads and stores.

Questa error reported for target/rv64gc/testlist.yaml, test: riscv_amo_test
```
# vsim -work /tools/riscv-dv/results/riscv-dv/riscv_amo_test/001/riscv-dv-build/work -c -do "/tools/riscv-dv/questa_sim.tcl" design_opt -sv_seed 3722723385 "+UVM_TESTNAME=riscv_rand_instr_test" "+num_of_tests=1" "+start_idx=0" "+asm_file_name=/tools/riscv-dv/results/riscv-dv/riscv_amo_test/001/riscv-dv-build/asm_tests/riscv_amo_test" -l /tools/riscv-dv/results/riscv-dv/riscv_amo_test/001/riscv-dv-build/sim_riscv_amo_test_0.log "+UVM_VERBOSITY=UVM_HIGH" "+instr_cnt=5000" "+num_of_sub_program=5" "+directed_instr_0=riscv_lr_sc_instr_stream,10" "+directed_instr_1=riscv_amo_instr_stream,10"
 ...
... <everything looks good here until Warning seen after riscv_instr_gen_config is randomized >
...

UVM_INFO verilog_src/questa_uvm_pkg-1.2/src/questa_uvm_pkg.sv(339) @ 0: reporter [Questa UVM] End Of Elaboration
# UVM_INFO /tools/riscv-dv/test/riscv_instr_test_lib.sv(27) @ 0: uvm_test_top [uvm_test_top] riscv_instr_gen_config is randomized:
# ---------------------------------------------------------------------------
# Name                               Type                    Size  Value
# ---------------------------------------------------------------------------
# cfg                                riscv_instr_gen_config  -     @1402
#   main_program_instr_cnt           integral                32    'h12b4
#   sub_program_instr_cnt            sa(integral)            5     -
#     [0]                            integral                32    'h8d1
#     [1]                            integral                32    'h201
#     [2]                            integral                32    'h3be
#     [3]                            integral                32    'h1c6
#     [4]                            integral                32    'h406
#   debug_program_instr_cnt          integral                32    'h305e3fd9
#   data_page_pattern                data_pattern_t          2     ALL_ZERO
#   init_privileged_mode             privileged_mode_t       2     USER_MODE
#   reserved_regs                    array(riscv_reg_t)      3     -
#     [0]                            riscv_reg_t             5     S10
#     [1]                            riscv_reg_t             5     S3
#     [2]                            riscv_reg_t             5     T0
#   ra                               riscv_reg_t             5     RA
#   sp                               riscv_reg_t             5     S3
#   tp                               riscv_reg_t             5     S10
#   tvec_alignment                   integral                32    'h2
#   no_data_page                     integral                1     'h0
#   no_branch_jump                   integral                1     'h0
#   no_load_store                    integral                1     'h0
#   no_csr_instr                     integral                1     'h0
#   no_ebreak                        integral                1     'h1
#   no_dret                          integral                1     'h1
#   no_fence                         integral                1     'h0
#   no_wfi                           integral                1     'h1
#   fix_sp                           integral                1     'h0
#   enable_unaligned_load_store      integral                1     'h0
#   illegal_instr_ratio              integral                32    'h0
#   hint_instr_ratio                 integral                32    'h0
#   boot_mode_opts                   string                  0     ""
#   enable_page_table_exception      integral                32    'h0
#   no_directed_instr                integral                1     'h0
#   enable_interrupt                 integral                1     'h0
#   enable_timer_irq                 integral                1     'h0
#   bare_program_mode                integral                1     'h0
#   enable_illegal_csr_instruction   integral                1     'h0
#   enable_access_invalid_csr_level  integral                1     'h0
#   enable_misaligned_instr          integral                1     'h0
#   enable_dummy_csr_write           integral                1     'h0
#   randomize_csr                    integral                1     'h0
#   allow_sfence_exception           integral                1     'h0
#   no_delegation                    integral                1     'h1
#   force_m_delegation               integral                1     'h0
#   force_s_delegation               integral                1     'h0
#   support_supervisor_mode          integral                1     'h1
#   disable_compressed_instr         integral                1     'h0
#   signature_addr                   integral                64    'hdeadbeef
#   num_of_harts                     integral                32    'h1
[... 81 lines cut ...]
UVM_INFO /tools/riscv-dv/src/isa/riscv_instr.sv(101) @ 0: reporter [riscv_instr] Registering VLHUFF_V
UVM_INFO /tools/riscv-dv/src/isa/riscv_instr.sv(101) @ 0: reporter [riscv_instr] Registering VLWUFF_V
UVM_INFO /tools/riscv-dv/src/isa/riscv_instr.sv(101) @ 0: reporter [riscv_instr] Registering VLEFF_V
UVM_INFO /tools/riscv-dv/src/isa/riscv_instr.sv(101) @ 0: reporter [riscv_instr] Registering CUSTOM_1
UVM_INFO @ 0: reporter [RNTST] Running test riscv_rand_instr_test...
UVM_INFO /tools/riscv-dv/test/riscv_instr_base_test.sv(42) @ 0: uvm_test_top [uvm_test_top] Create configuration instance
UVM_INFO /tools/riscv-dv/src/riscv_instr_gen_config.sv(582) @ 0: reporter [cfg] riscv_instr_pkg::supported_privileged_mode = 3
UVM_INFO/tools/riscv-dv/src/riscv_pmp_cfg.sv(133) @ 0: reporter [pmp_cfg] pmp max offset: 0xffffffffffffffff
UVM_INFO /tools/riscv-dv/src/riscv_pmp_cfg.sv(154) @ 0: reporter [pmp_cfg] MAX OFFSET: 0xffffffffffffffff
UVM_INFO /tools/riscv-dv/test/riscv_instr_base_test.sv(44) @ 0: uvm_test_top [uvm_test_top] Create configuration instance...done
  rand pmp_cfg_reg_t pmp_cfg[];
                           |
xmsim: *E,RNDCNSTE (./src/riscv_pmp_cfg.sv,43|27): Randomization constraint has this error, which will cause the randomize function to return 0 and no new rand values will be set:
rand unpacked structs are not currently supported; use "-xceligen rand_struct" to enable their use.
    `DV_CHECK_RANDOMIZE_FATAL(cfg)
                                 |
xmsim: *W,SVRNDF (./test/riscv_instr_test_lib.sv,26|33): The randomize method call failed. The unique id of the failed randomize call is 0.
Observed simulation time : 0 FS + 44
UVM_FATAL /tools/riscv-dv/test/riscv_instr_test_lib.sv(26) @ 0: uvm_test_top [uvm_test_top] Check failed (cfg.randomize()) Randomization failed! 
````

Am I missing something in the simulator options?

Thanks,
Vandana

vandanaprabhu, 2020-07-07:
BTW, there is a bug in the definition of PMPCFG regsiters `src/riscv_privil_reg.sv` : 
v1.10 of the privileged spec says RV64 uses pmpcfg2 not pmpcfg1. 
```
The PMP configuration registers are densely packed into CSRs to minimize context-switch time.
For RV32, four CSRs, pmpcfg0–pmpcfg3, hold the configurations pmp0cfg–pmp15cfg for the 16
PMP entries, as shown in Figure 3.27. For RV64, pmpcfg0 and pmpcfg2 hold the configurations
for the 16 PMP entries, as shown in Figure 3.28; pmpcfg1 and pmpcfg3 are illegal.
RV64 systems use pmpcfg2, rather than pmpcfg1, to hold configurations for PMP entries 8–15.
This design reduces the cost of supporting multiple MXLEN values, since the configurations for
PMP entries 8–11 appear in pmpcfg2[31:0] for both RV32 and RV64.
```
The correct definition should be:
```
     // Physical Memory Protection Configuration Register
      PMPCFG0: begin
        privil_level = M_LEVEL;
        add_field("PMP0CFG", 8, WARL);
        add_field("PMP1CFG", 8, WARL);
        add_field("PMP2CFG", 8, WARL);
        add_field("PMP3CFG", 8, WARL);
        if(XLEN==64) begin
          add_field("PMP4CFG", 8, WARL);
          add_field("PMP5CFG", 8, WARL);
          add_field("PMP6CFG", 8, WARL);
          add_field("PMP7CFG", 8, WARL);
        end
      end
      // Physical Memory Protection Configuration Register
      PMPCFG2: begin
        privil_level = M_LEVEL;
        if(XLEN==64) begin
          add_field("PMP8CFG", 8, WARL);
          add_field("PMP9CFG", 8, WARL);
          add_field("PMP10CFG", 8, WARL);
          add_field("PMP11CFG", 8, WARL);
          add_field("PMP12CFG", 8, WARL);
          add_field("PMP13CFG", 8, WARL);
          add_field("PMP14CFG", 8, WARL);
          add_field("PMP15CFG", 8, WARL);
       end else begin
          add_field("PMP8CFG", 8, WARL);
          add_field("PMP9CFG", 8, WARL);
          add_field("PMP10CFG", 8, WARL);
          add_field("PMP11CFG", 8, WARL);
        end
      end
      // Physical Memory Protection Configuration Register
      PMPCFG1: begin
        if(XLEN!=32) begin
          `uvm_fatal(get_full_name(), "CSR PMPCFG1 only exists in RV32.")
        end else begin
        privil_level = M_LEVEL;
        add_field("PMP4CFG", 8, WARL);
        add_field("PMP5CFG", 8, WARL);
        add_field("PMP6CFG", 8, WARL);
        add_field("PMP7CFG", 8, WARL);
        end
      end
      // Physical Memory Protection Configuration Register
      PMPCFG3: begin
        if(XLEN!=32) begin
          `uvm_fatal(get_full_name(), "CSR PMPCFG3 only exists in RV32.")
        end else begin
        privil_level = M_LEVEL;
        add_field("PMP12CFG", 8, WARL);
        add_field("PMP13CFG", 8, WARL);
        add_field("PMP14CFG", 8, WARL);
        add_field("PMP15CFG", 8, WARL);
        end
      end

```

taoliug, 2020-07-08:
PMP + virtual address translation is currently not supported.

vandanaprabhu, 2020-07-08:
Do you have any plans for this? Is there a reason not to support this?
It would have been good to document this somewhere as I didn't see this mentioned in the code or otherwise. Better still if the compile log flags this as unsupported else the randomization fails in bizarre ways.

udinator, 2020-07-27:
Hi @vandanaprabhu yes eventually we want to support PMP with virtual address translation, it just has not been done as of right now.

Thanks for catching the bug regarding `src/riscv_privil_reg.sv`, I'll get a patch ready for it.