chipsalliance/riscv-dv #641: [Pygen] Issue with imm value and shift operation

aneels3, 2020-07-11:
Immediate values are not properly generated because of the issue with shift operation in **extend_imm()** function of **riscv_instr.py**. 
[image]

As I am considering `self.imm = vsc.rand_uint32_t()` type for randomizing the imm value.
Facing compatibility issue while doing or operation with self.imm_mask and self.imm

If I declared self.imm_mask as PyVSC data type then it is difficult to assign its value default to `0xffffffff` 

To solve the above issue. I have declared the imm_mask value directly in the constructor function. The imm_mask size increases after the shift operation to more than 32 bits. So to limit its size to 32 bits I did the bitwise & operator with `0xffffffff` while shifting.
I believe there must be some other way to solve this issue.
Need some suggestion on this.

hodjat91, 2020-07-12:
Hi @aneels3 
I might miss something here, but if you declare self.imm_mask as a non-random PyVSC data type (say `vsc.uint32_t` or `vsc.bit_t(32)` ), what's the problem with assigning it a default value of `0xffffffff`? I believe that you can do so either by passing the default value while you define it `(self.imm_mask = vsc.uint32_t(the_default_value))` or later on using `set_val()` method.

aneels3, 2020-07-13:
Hi @hodjat91 
Thanks for the suggestion, I completely miss out the default assignment part that you have suggested `(self.imm_mask = vsc.uint32_t(the_default_value))`
It works for me now.

feel free to close this issue upon the successful merging of the #628

hodjat91, 2020-07-13:
Hi @aneels3 
Sounds great!

closed 2020-07-13 by hodjat91