chipsalliance/riscv-dv #642: [Pygen] None immediate value in asm file

aneels3, 2020-07-11:
Getting **None** imm value in asm file for the first randomized immediate instr
[image]

You can refer the **srli** instruction for a better understanding of the issue.

hodjat91, 2020-07-12:
Hi @aneels3 
I don't think there is a need for defining `self.imm_str` as a vsc data type (makes more sense to define `self.imm` as a random vsc data type, apply the required constraints on it, and then just cast it to str, as I believe you already did so). Could you please double check and make sure that you don't have constraints that are in conflict with each other (cause if there are such constraints, then this 'none' value makes sense, although I don't believe that this is the case). So alternatively, could you please generate several other programs and see whether or not this behavior is consistent across all of them? I am a bit suspicious of the functionality of the PyVSC randomizer. It might worth looking at the vsc randomizer source code and see how it exactly works. Getting 'none' result in some cases might be true if randomizer tries to find a solution within a default (and limited) number of steps, so when it fails, 'none' will be returned. One of the solvers offered in `python-constraint` module exploits similar mechanism.

mballance, 2020-07-12:
Hi @aneels3  and @hodjat91,
With respect to PyVSC randomization, you should be seeing exceptions thrown if the constraints cannot be resolved. It sounds like you're not seeing this. 
How is the string value of the immediate field formed? I'm not aware of cases where PyVSC is intentionally setting field values to Python 'None'.

-Matthew

hodjat91, 2020-07-13:
Hi @mballance ,

Thank you for the clarification on the PyVSC randomization process. The `self.imm_str` is nothing but a string cast of the randomized immediate value in instructions that operate on immediate values `(i.e., addi).` The randomized immediate value itself `(self.imm)` is a standard-width random data type of PyVSC. 

-Hodjat

aneels3, 2020-07-13:
Hi @hodjat91 and @mballance 
I don't think this is something related to randomization failure.
I forgot to mention that I have declared `self.imm_str = None` in the constructor function of the base class.
I cross-checked by assigning some other default value in the constructor function, in this case, I am getting default assigned value instead if **None**.

-Anil

hodjat91, 2020-07-13:
Hi @aneels3 
Thanks for the update. So in this case, my assumption is that the `get_imm()` and/or `update_imm_str()` are not called properly. Could you please double-check and make sure that `self.imm` is actually randomized when you see None/default assigned value for the `self.imm_str`?

-Hodjat

hodjat91, 2020-07-14:
Hi @aneels3 
I looked at your code and found a weird behavior: In `class riscv_instr`, instead of using the `post_randomize(self)` (and hence rely on the PyVSC randomizer to take care of the post randomization process), if you decide to define an explicit method (say `post_process(self)`) and call this method after randomizing your `riscv_instr_ins` object, the "None" issue in instructions operating on immediate values gets resolved. Initially, I thought that the `post_randomize()` method might not get called properly, but I actually created a similar scenario and it worked fine. Let's use this `post_process()` alternative until we find out where does the issue come from. (please add a side note for this to be fixed in future). 

-Hodjat

aneels3, 2020-07-14:
Hi @hodjat91 
That makes sense. I have tried as per your suggestion and its working fine now.
I have added a side note as well to map it in a post_randomize() method.
Thanks again!

-Anil

hodjat91, 2020-07-19:
Update: This was an issue with the `post_randomize()` process in the PyVSC library, which is resolved in the new release.

closed 2020-07-19 by hodjat91