chipsalliance/riscv-dv #643: Compilation error for the rv32i target - MAX_LMUL identifier undeclared

dariuszst, 2020-07-13:
The compilation of riscv-dv files for rv32i target ends with an error:

`ERROR VCP5103 "Undeclared identifier: MAX_LMUL." "/home/dariuszst/riscv-dv_latest/src/riscv_vector_cfg.sv" 75  28
`

This is caused by the new constraint added in commit "Add MAX LMUL configure (#609)":
https://github.com/google/riscv-dv/blob/17eec2d00c6f1d162016902806c73bdbeec6f566/src/riscv_vector_cfg.sv#L73-L75

That uses "MAX_LMUL" parameter that is defined in riscv_core_setting.sv file for all target except rv32i.
Lack of this declaration causes compilation error.

udinator, 2020-07-17:
Hi @dariuszst thanks for bringing this up!
I'll push a PR to fix this compile error.

closed 2020-07-17 by udinator

dariuszst, 2020-07-20:
OK. Thanks for a fix.