chipsalliance/riscv-dv #644: Mismatch between custom CSR model and riscv standards

jiangxuanfeng, 2020-07-14:
I want to verify my RTL with instruction generator. But my CSR model is not consistent with the standard riscv. For example, I don't have mhrtid and misa. The address of the custom mepc is inconsistent with the standard riscv. Can I modify the address and properties of CSR to match my RTL? which file should I modify?

udinator, 2020-07-17:
Hi @jiangxuanfeng thanks for bringing this up.
Right now, riscv-dv doesn't support a custom CSR model (different addresses of riscv CSRs, and custom CSRs), this is something that I will be working to implement soon. 

With regards to your RTL not supporting `mhartid` and `misa` - these are two CSRs that the riscv-dv generator assumes to exist as it is running, as above I'll keep this request in mind when implementing custom CSRs.

udinator, 2020-07-23:
Hi @jiangxuanfeng - just an update regarding your request about custom CSRs.
Our plan for riscv-dv is to implement support for fully custom CSRs (e.g. non-standard CSRs defined by an RTL implementation).
We don't plan to support either the exclusion of CSRs deemed required by the official spec or modifying the addresses of standard CSRs.

However, there is something you can do locally if you want to modify the addresses of standard CSRs. In `src/riscv_instr_pkg.sv`, there is an enum `privileged_reg_t` which provides name:address mappings for all standard CSRs - for your scenario, you can change this defined address for the `mepc` CSR, and this should solve that issue.

For your scenario of the RTL not having `mhartid` or `misa`, you can go to the `riscv_core_setting.sv` file for whichever target you are using, and delete those two CSRs from the `implemented_csrs` list there. However, be aware that doing this will cause some issues as a good part of the generated initialization routine uses mhartid and misa, so you will have to go and locally prune these code sections.

udinator, 2020-08-05:
Hi @jiangxuanfeng - I'm going to go ahead and close out this issue, please feel free to either reopen it or open a new issue if you have any other questions!

closed 2020-08-05 by udinator