chipsalliance/riscv-dv #647: [Pygen] Issue with instruction name in constraint

ShraddhaDevaiya, 2020-07-16:
Hello, we are trying to implement **imm_c** constraint with vsc library in **riscv_instr.py** file. But we are facing issue for **instr_name** in **imm_c** constraint. Issue is  we are not getting **instr_name** in **imm_c** constraint, while if we check same **self.instr_name** in **post_randomize** method , in that we are getting **instr_name** value. 

[image]

Then we tried to print the object which we are passing to **imm_c constraint** and **post_randomize**. In this we can see that object id is same for particular instruction in both functions. But instr_name we are not getting in **imm_c**. Following is the log for this :

[name.log](https://github.com/google/riscv-dv/files/4929772/name.log)
In this you can check **object id** and **instr_name** for particular instruction.

So, can anyone please help me in this issue?

hodjat91, 2020-07-19:
Hi @ShraddhaDevaiya 
I think this makes sense, and here is the reason why you're facing this behavior: `self.instr_name` itself is a `vsc.rand_sth`. Every method with a `@vsc.constraint` decorator is implicitly called while you instantiate an object. With that said, it makes sense not to get a proper value for `self.instr_name` inside of the constraints (as the object has not been "randomized" yet). The values for any `vsc.rand_sth` variable should be known only after the point where `object.randomize()` is called (and that's why you can see the value for `self.instr_name` in the `post_randomize()` method as it is called after the randomization process). So don't worry about not getting a proper value for a random variable within the methods with `@vsc.constraint` decorator, just define your constraints and you should get valid data (according to the constraints) after randomization process. 

[Here](https://github.com/google/riscv-dv/files/4943304/temp.log) is a small program that illustrates what I mentioned. Please let me know if anything is unclear.

ShraddhaDevaiya, 2020-07-20:
Okay got the idea. Thanks for helping !

ShraddhaDevaiya, 2020-07-20:
Hello @hodjat91, I have implemented imm_c contraint in riscv_instr.py file. 

[image]

But while running the program, if we check log, it is not printing "INSIDE IF"  and I have checked assembly file. We are getting  these **slli, srli and srai** instructions, but in **imm_c** constraint it is not going inside **if loop** . For this probably the reason is , we are not getting **instr_name** in **imm_c**.
Here is the generated log file : 
[logname.log](https://github.com/google/riscv-dv/files/4946762/logname.log)

So  can you please suggest, what can be the problem ?

hodjat91, 2020-07-20:
Hi @ShraddhaDevaiya ,

The reason behind not entering inside of the second `if` statement is that at this point we don't know what's the **_randomly_** generated value for `riscv_instr_name`. For the constraint methods (decorated with `vsc.constraint`), the syntax for the conditional statements (like an `if` statement) operating on a `vsc.rand_sth` variable is a bit different than what you wrote (it's actually pretty similar to that of SystemVerilog). For example, if `self.imm[5:11]` should be zero in case of instruction name = `slli`, you could do it this way:

```
with vsc.if_then(self.instr_name == riscv_instr_name_t.SLLI.name): 
    self.imm[5:11] == 1
```

You can find more examples [here](https://py-vsc.readthedocs.io/en/latest/constraints.html). Please let me know if the issue still persists.

-Hodjat

ShraddhaDevaiya, 2020-07-20:
Hi @hodjat91 , I tried to implement constraint  with **vsc.if_then** , but it is giving an error like following : 

[image]

Finding the cause for this. can you please help in this?

hodjat91, 2020-07-21:
Hi @ShraddhaDevaiya 

Could you please share the code with me so I can get a better understanding of the issue? Thanks.

- Hodjat

ShraddhaDevaiya, 2020-07-21:
Yeah Sorry for that. 
Here is the code :
```
@vsc.constraint
    def imm_c(self):
        with vsc.if_then(self.instr_name == riscv_instr_name_t.SLLI.name):
            print("INSIDE")
            self.imm[5:11] == 0

        with vsc.if_then(self.instr_name == riscv_instr_name_t.SRAI.name):
            self.imm[5:11] == 0

        with vsc.if_then(self.instr_name == riscv_instr_name_t.SRLI.name):
            self.imm[5:11] == 0

```
Thanks.
-Shraddha

hodjat91, 2020-07-21:
Hi @ShraddhaDevaiya 

I looked at your code, here is the issue with it: 
Assuming that you defined `self.instr_name` as a vsc random variable (e.g., `self.instr_name = vsc.rand_enum_t(name_of_enum_class)`): there are two issues in your code, first is the `.name` at the end of instruction name (in your `with vsc.if_then` statement), you do not need `.name` extension. Next issue is with the part select constraint (`self.imm[5:11]`) that actually should be `self.imm[11:5]`. 

[Here](https://github.com/google/riscv-dv/files/4951837/test.py.log) is a sample program that does what you want. Please let me know if you have further questions.

ShraddhaDevaiya, 2020-07-21:
Hi @hodjat91 , it is working now. Thanks for helping !

closed 2020-07-21 by ShraddhaDevaiya