chipsalliance/riscv-dv #669: function coverage

yinhanquan, 2020-08-04:
Function coverage run with ovpsim , like :cov --dir out/ovpsim_sim --iss ovpsim --target rv32imc ,but there is no coverage data,we also try run with "--cov" first enable function coverage ,get the same result .

hodjat91, 2020-08-04:
Hi Yinhanquan,

I believe `--cov` flag doesn't have an effect as of now. After you run `cov.py` using a command like: `cov --dir out_2020-08-03/ovpsim_sim/ --iss ovpsim`, you should get an info message that says: `Coverage results are saved to cov_out_2020-08-03` if building the coverage framework finishes successfully. Then, say that you want to see the coverage report in a textual format, you can get it by running: `urg -full64 -dir test.vdb/ -format text` (assuming you have vcs) from your coverage results directory (final coverage report will be saved to `urgReport` directory).

yinhanquan, 2020-08-04:
thank you ,i have sloved this question.

closed 2020-08-04 by yinhanquan