chipsalliance/riscv-dv #671: RISCV-DV compile issues

sfjiang-astri, 2020-08-05:
Hi All,

when i compile the RISCV-DV, serveral errors occured. All the files are download from the repo,and the error looks like following:
    if (sub_extension == "zvlsseg") {
                                 |
ncvlog: *E,RNCTYP (/home/sfjiang/RISCV-DV/riscv-dv/src/isa/riscv_vector_instr.sv,186|33): A 'string type' cannot be used within a constraint expression.

sfjiang-astri, 2020-08-05:
another error : i'm sure all the sv file are included in the testfile, while the compiler cannot find the class header.
riscv_instr_gen_config  cfg;
                       |
ncvlog: *E,NOIPRT (/home/sfjiang/RISCV-DV/riscv-dv/test/riscv_instr_base_test.sv,21|23): Unrecognized declaration 'riscv_instr_gen_config' co

udinator, 2020-08-05:
Hi @sfjiang-astri - I believe the error you are mentioning is stale, it has been fixed by a recent commit.
It looks like you are running with a Cadence simulator - I've tried compiling the current top of tree with both Xcelium and IUS, and no errors are resulting.
Can you please sync to the top of tree and try to compile again?

sfjiang-astri, 2020-08-06:
> Hi @sfjiang-astri - I believe the error you are mentioning is stale, it has been fixed by a recent commit.
> It looks like you are running with a Cadence simulator - I've tried compiling the current [...]

Thanks for your reply. I used latest version and it said -->
FileNotFoundError: [Errno 2] No such file or directory: '/home/sfjiang/RISCV-DV/SomeLocalDir/bin/riscv32-unknown-elf-gcc': '/home/sfjiang/RISCV-DV/SomeLocalDir/bin/riscv32-unknown-elf-gcc'
is that means my environment has some error, or some file lost?

hodjat91, 2020-08-06:
Hi @sfjiang-astri 

Not being able to find riscv32-unknown-elf-gcc means that the riscv-gcc compiler toolchain has not been properly installed on your machine. Please follow the instructions from [here](https://htmlpreview.github.io/?https://github.com/google/riscv-dv/blob/master/docs/build/singlehtml/index.html#document-index) to install it.

sfjiang-astri, 2020-08-06:
> Hi @sfjiang-astri
>
> Not being able to find riscv32-unknown-elf-gcc means that the riscv-gcc compiler toolchain has not been properly installed on your machine. Please follow the instructions from [...]

thanks hodjat91 ~
shoulud I instll both gcc and iss together within your link?

hodjat91, 2020-08-06:
No problem. Yes, you should be able to install both using instructions provided in the provided link. 

- Hodjat

sfjiang-astri, 2020-08-07:
> No problem. Yes, you should be able to install both using instructions provided in the provided link.
> 
> -Hodjat

Hi Hodjat,
I installed both riscv-gcc and spike and then run the test using IUS. the error occurs like : 
"ERROR return code: 1/1, cmd:/opt/riscv32/bin/spike --log-commits --isa=rv32imc -l out_2020-08-07/asm_tests/riscv_arithmetic_basic_test_0.o &> out_2020-08-07/spike_sim/riscv_arithmetic_basic_test.0.log"
inside the log file:
"/opt/riscv32/bin/spike: /lib64/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /opt/riscv32/bin/spike)
/opt/riscv32/bin/spike: /lib64/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /opt/riscv32/bin/spike)
/opt/riscv32/bin/spike: /lib64/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /opt/riscv32/bin/spike)"
is that means I need to install mising file, or something wrong when i install spike?
thx

hodjat91, 2020-08-07:
Hi @sfjiang-astri 

As you can see in the message from the log file, it seems like the Spike is missing some of its prerequisites (the `GLIBCXX_version` is actually needed by Spike). So I would suggest either install all the prerequisites and required libraries for Spike, or try another ISS (I'd suggest riscv-ovpsim as the installation process is pretty straightforward).

-Hodjat

hodjat91, 2020-08-11:
Hi @sfjiang-astri 

Is your issue with the compilation process fixed now?

sfjiang-astri, 2020-08-11:
> Hi @sfjiang-astri
> 
> Is your issue with the compilation process fixed now?

Hi Hodjat,

Thanks for your help. the problem solved using ovpsim.

sfjiang-astri, 2020-08-11:
> Hi @sfjiang-astri
> 
> Is your issue with the compilation process fixed now?

Hi Hodjat,

I still have some basic questions on RISCV-DV.
Is there a DUT in this dv testbench?
Does this riscv-dv has any relationship with rocket chips?
How can i dump a waveform for the sample tests?

Thanks

udinator, 2020-08-11:
> Hi Hodjat,
> 
> I still have some basic questions on RISCV-DV.
> Is there a DUT in this dv testbench?
> Does this riscv-dv has any relationship with rocket chips?
> How can i dump a waveform for the sample tests?
> 
> Thanks 

Hi @sfjiang-astri,

This DV testbench does not have any DUT, as this codebase is meant solely to generate output random RISC-V assembly test files for use by an external DUT.
Consequently, this environment does not support dumping waveforms.
If you are interested in integrating this tool to a DV environment for some RISC-V core, I suggest looking at https://github.com/lowRISC/ibex/tree/master/dv/uvm/core_ibex as a workable example.

This is also a separate RISC-V project from the Rocket chip generator, which is written in Chisel and allows you to generate various configurations of the Rocket core.

Best,
Udi

sfjiang-astri, 2020-08-13:
> Hi @sfjiang-astri,
>
> This DV testbench does not have any DUT, as this codebase is meant solely to generate output random RISC-V assembly test files for use by an external DUT.
> Consequently, this [...]

Hi Udi,
is it possible to use this riscv-dv environment to set up a verification testbench for a riscv project (rocket chip)?
And how to use it (the role it plays in the verification)?

sfjiang-astri, 2020-08-14:
> https://github.com/lowRISC/ibex/tree/master/dv/uvm/core_ibex

Hi,
for this example, https://github.com/lowRISC/ibex/tree/master/dv/uvm/core_ibex,
How can i dunp the waveform? Which command can i use to run the tests?
thanks

udinator, 2020-08-14:
Hi @sfjiang-astri - yes it is very possible to set up a verification testbench for an RTL core that integrates the riscv-dv generator.

In the [core_ibex](https://github.com/lowRISC/ibex/tree/master/dv/uvm/core_ibex) environment, we have a Makefile that controls the entire simulation flow.  We also need a Python script `sim.py` to control the compilation and simulation of the DV environment.
On top of this, we also need to create a `riscv_dv_extension` directory to provide a YAML testlist and extend instruction generator classes as needed - the testlist is located at `riscv_dv_extension/testlist.yaml` and provides YAML definitions of each test that the simulation flow will parse and use as necessary in each step of the flow (e.g. the options listed under the `gen_opts` section of each test is used to control instruction generation, the options listed under `sim_opts` section of the YAML test are used to control parameters for DV simulation, etc...).
Also, while creating the UVM testbench, it is necessary to add functionality in the base test class (as we do in `tests/core_ibex_base_test.sv`) to load the compiled binary program into the memory model of the testbench, this way the core is then able to jump to the start of the random program at the beginning of simulation time.

To directly address the question you asked about what role riscv-dv plays in verification - riscv-dv generates random programs with various parameter configurations that will test different parts of the core (e.g. pipeline hazards, stressing MMU, PMP, exceptions, debug mode, etc...).

To actually run simulations in the Ibex environment specifically, the general format looks something like this:
```console
make ITERATIONS=<num_iterations> SEED=<seed_num> ISS=<iss_name> TEST=<test_name> OUT=<output_dir>
```
To dump waves, you can append `WAVES=1` to the Make command and to enable coverage you can append `COV=1` to the Make command - there are also several other options for the Make command that you can play with, these are all listed in the Makefile.

For example, if you want to run one iteration of the basic arithmetic instruction test with the Spike ISS, enabling waveforms and coverage, you would run:
```console
make TEST=riscv_arithmetic_basic_test ISS=spike ITERATIONS=1 WAVES=1 COV=1
```

Some notes:
- `<output_dir>` defaults to `out/`
- `test_name` must match the name of one of the tests defined in `riscv_dv_extension/testlist.yaml`.
- `iss_name` must be either "ovpsim" or "spike", "ovpsim" is used by default.
- The output will be generated in `<output_dir>/` - logs for the instruction generation for this test is found in `<output_dir>/seed-<seed_num>/instr_gen/*`, while logs for the DV simulation will be found in `<output_dir>/seed-<seed_num>/rtl_sim/*`

Hope this helps!

sfjiang-astri, 2020-08-14:
Hi Udi,

Thanks for your reply. its very helpful to me.
i run the following cammand: 

"make SIMULATOR=ius TEST=riscv_arithmetic_basic_test ISS=ovpsim ISA=rv32imc ITERATIONS=1 OUT=out_test2 Waves=1"

you can see that i wanna use ius (irun simulator) to perform the simulation. I added ius option into sim_makefrag_gen.py and ibex_config.py (adapted from the "questa" option). And the error looks like : 

[image]

Should I modify more configure files?

Thanks 
sfjiang

udinator, 2020-08-14:
Hi @sfjiang-astri we actually do not support IUS in the Ibex environment due to some simulator limitations that prevent compilation.

Full disclaimer - you are welcome to try and integrate IUS locally and modify the RTL to allow support, however this will likely introduce errors that you will need to fix yourself as we will not support it in Ibex (I suggest using another simulator if you are able to, this will prevent any problems from arising).

The error you are seeing `Error: Cannot find RTL simulator ius` is coming from the file `yaml/rtl_simulation.yaml` - if you want to move forward with IUS, you will have to add a new target for the ius simulator, you can use the other simulator targets already in this file for reference.

sfjiang-astri, 2020-08-17:
Thanks Udi.
I tried IUS but filed since so many errors occured when integrated IUS in the environment. 
Actually, i dont understand the testbench structure of this ibex,  in which file and in what hierarchy the DUT is instantiated.
and how to instantiate the DUT with the RISCV-DV? 
thanks ~

sfjiang-astri, 2020-08-18:
> Hi @sfjiang-astri we actually do not support IUS in the Ibex environment due to some simulator limitations that prevent compilation.
>
> Full disclaimer - you are welcome to try and integrate IUS [...]

udinator, 2020-08-18:
Hi @sfjiang-astri - apologies for my delayed response!

For the Ibex repo, the top level testbench where the DUT is instantiated is found at `tb/core_ibex_tb_top.sv` - we instantiate the `ibex_core_tracing` module to gain access to the RTL tracer, it is a pretty standard testbench top file so we just set up all the interfaces and place them all into `uvm_config_db`.
The main interface agents we use (to model IMEM and DMEM) are found in `common/ibex_mem_intf_agent/*` - note that we only make use of the `response` agents as DMEM/IMEM will only respond to memory requests that the core initiates.

I'm not quite sure I understand your question `how to instantiate the DUT with the RISCV-DV` - the RISCV-DV instruction generator actually has no part in instantiating the Ibex DUT. In the Ibex simulation flow, the RISCV-DV is just used to generate some random assembly test files, and then we invoke GCC to compile them into binaries.
If you take a look at the `load_binary_to_mem()` function in `tests/core_ibex_base_test.sv`, you will see that it takes in a file pathname from a `+bin` plusarg, this is assumed to be the pathname of the compiled binary test file. We then read in this binary file one byte at a time and write it to the memory model that we instantiate in the testbench (the memory model itself can be found at `../../../vendor/lowrisc_ip/mem_model/mem_model.sv`, assuming  that you are starting from the `dv/uvm/core_ibex` directory).

closed 2020-08-27 by udinator