chipsalliance/riscv-dv #681: Instruction generator not working

aliess, 2020-08-18:
Hello everyone,

I was told to open an issue here, I opened an issue on ibex here first https://github.com/lowRISC/ibex/issues/1071.

The instruction generator doesn't seem to be working when using questa with any test other than the riscv_arithmetic_basic_test.

I cloned this repo and used 
`python3 run.py --test riscv_arithmetic_basic_test --simulator questa`

and I get the following log:
```
Tue, 18 Aug 2020 12:18:33 INFO     Creating output directory: out_2020-08-18
Tue, 18 Aug 2020 12:18:33 INFO     Processing regression test list : /home/ali/Documents/riscv-dv/target/rv32imc/testlist.yaml, test: riscv_arithmetic_basic_test
Tue, 18 Aug 2020 12:18:33 INFO     Processing regression test list : /home/ali/Documents/riscv-dv/yaml/base_testlist.yaml, test: riscv_arithmetic_basic_test
Tue, 18 Aug 2020 12:18:33 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:2
Tue, 18 Aug 2020 12:18:33 INFO     Processing simulator setup file : /home/ali/Documents/riscv-dv/yaml/simulator.yaml
Tue, 18 Aug 2020 12:18:33 INFO     Found matching simulator: questa
Tue, 18 Aug 2020 12:18:33 INFO     Building RISC-V instruction generator
Tue, 18 Aug 2020 12:18:46 INFO     Running RISC-V instruction generator
Tue, 18 Aug 2020 12:18:46 INFO     Generating 2 riscv_arithmetic_basic_test
Tue, 18 Aug 2020 12:18:46 INFO     Running riscv_arithmetic_basic_test with 1 batches
Tue, 18 Aug 2020 12:18:46 INFO     Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Tue, 18 Aug 2020 12:18:55 INFO     Compiling out_2020-08-18/asm_tests/riscv_arithmetic_basic_test_0.S
Tue, 18 Aug 2020 12:18:55 INFO     Converting to out_2020-08-18/asm_tests/riscv_arithmetic_basic_test_0.bin
Tue, 18 Aug 2020 12:18:55 INFO     Compiling out_2020-08-18/asm_tests/riscv_arithmetic_basic_test_1.S
Tue, 18 Aug 2020 12:18:55 INFO     Converting to out_2020-08-18/asm_tests/riscv_arithmetic_basic_test_1.bin
Tue, 18 Aug 2020 12:18:55 INFO     Processing ISS setup file : /home/ali/Documents/riscv-dv/yaml/iss.yaml
Tue, 18 Aug 2020 12:18:55 INFO     Found matching ISS: spike
Tue, 18 Aug 2020 12:18:55 INFO     spike sim log dir: out_2020-08-18/spike_sim
Tue, 18 Aug 2020 12:18:55 INFO     Running spike sim: out_2020-08-18/asm_tests/riscv_arithmetic_basic_test_0.o
Tue, 18 Aug 2020 12:19:05 ERROR    Timeout[10s]: /home/ali/Documents/riscv-isa-sim/build/spike --log-commits --isa=rv32imc -l out_2020-08-18/asm_tests/riscv_arithmetic_basic_test_0.o &> out_2020-08-18/spike_sim/riscv_arithmetic_basic_test.0.log
Tue, 18 Aug 2020 12:19:05 INFO     Running spike sim: out_2020-08-18/asm_tests/riscv_arithmetic_basic_test_1.o
```

while using another test results in the following

`python3 run.py --test riscv_rand_instr_test --simulator questa`

```
Tue, 18 Aug 2020 12:19:46 INFO     Creating output directory: out_2020-08-18
Tue, 18 Aug 2020 12:19:46 INFO     Processing regression test list : /home/ali/Documents/riscv-dv/target/rv32imc/testlist.yaml, test: riscv_rand_instr_test
Tue, 18 Aug 2020 12:19:46 INFO     Processing regression test list : /home/ali/Documents/riscv-dv/yaml/base_testlist.yaml, test: riscv_rand_instr_test
Tue, 18 Aug 2020 12:19:46 INFO     Found matched tests: riscv_rand_instr_test, iterations:2
Tue, 18 Aug 2020 12:19:46 INFO     Processing simulator setup file : /home/ali/Documents/riscv-dv/yaml/simulator.yaml
Tue, 18 Aug 2020 12:19:46 INFO     Found matching simulator: questa
Tue, 18 Aug 2020 12:19:46 INFO     Building RISC-V instruction generator
Tue, 18 Aug 2020 12:19:52 INFO     Running RISC-V instruction generator
Tue, 18 Aug 2020 12:19:52 INFO     Generating 2 riscv_rand_instr_test
Tue, 18 Aug 2020 12:19:52 INFO     Running riscv_rand_instr_test with 1 batches
Tue, 18 Aug 2020 12:19:52 INFO     Running riscv_rand_instr_test, batch 1/1, test_cnt:2
Tue, 18 Aug 2020 12:20:00 ERROR    Cannot find assembly test: out_2020-08-18/asm_tests/riscv_rand_instr_test_0.S
```

aliess, 2020-08-18:
One more thing to add, I had to modify the dv_defines.svh file for this to run
```
`ifndef DV_CHECK
`define DV_CHECK(T_, MSG_="", SEV_=error, ID_=`gfn, WITH_C_=) \
    if (!(T_ WITH_C_)) begin \
      `uvm_``SEV_(ID_, $sformatf("Check failed (%s) %s ", `"T_`", MSG_)); \
    end
`endif

`ifndef DV_CHECK_FATAL
  `define DV_CHECK_FATAL(T_, MSG_="", ID_=`gfn, WITH_C_=) \
    `DV_CHECK(T_, MSG_, fatal, ID_, WITH_C_)
`endif
```

modified to remove `WITH_C_` argument

```
`ifndef DV_CHECK
`define DV_CHECK(T_, MSG_="", SEV_=error, ID_=`gfn) \
    if (!(T_)) begin \
      `uvm_``SEV_(ID_, $sformatf("Check failed (%s) %s ", `"T_`", MSG_)); \
    end
`endif

`ifndef DV_CHECK_FATAL
  `define DV_CHECK_FATAL(T_, MSG_="", ID_=`gfn) \
    `DV_CHECK(T_, MSG_, fatal, ID_)
`endif
```

or else it would complain the following:

```
Expansion of macro 'DV_CHECK_FATAL' failed: The number of actual arguments (3) are not equal to the number of formal arguments (4).
```

udinator, 2020-08-18:
Hi @aliess thanks for opening the issue here!

Just to confirm, are you seeing these errors with the latest commits on the upstream branch?

@danghai do you see any of these errors when you run with Questasim?

aliess, 2020-08-21:
do you mean the master branch ? if yes then yes I do have the latest one.

taoliug, 2020-08-26:
Can you share the simulation log and generated assembly tests?

danghai, 2020-08-29:
Hi, sorry for my late response. It works for me for both `Questa` and `Qrun`. I use version 2020.2 :
```
danghai@ubuntu:~/hhoangda/riscv-dv$ python3 run.py --test riscv_rand_instr_test --simulator questa
Fri, 28 Aug 2020 18:07:06 INFO     Creating output directory: out_2020-08-28
Fri, 28 Aug 2020 18:07:06 INFO     Processing regression test list : /home/danghai/hhoangda/riscv-dv/target/rv32imc/testlist.yaml, test: riscv_rand_instr_test
Fri, 28 Aug 2020 18:07:06 INFO     Processing regression test list : /home/danghai/hhoangda/riscv-dv/yaml/base_testlist.yaml, test: riscv_rand_instr_test
Fri, 28 Aug 2020 18:07:06 INFO     Found matched tests: riscv_rand_instr_test, iterations:2
Fri, 28 Aug 2020 18:07:06 INFO     Processing simulator setup file : /home/danghai/hhoangda/riscv-dv/yaml/simulator.yaml
Fri, 28 Aug 2020 18:07:06 INFO     Found matching simulator: questa
Fri, 28 Aug 2020 18:07:06 INFO     Building RISC-V instruction generator
Fri, 28 Aug 2020 18:07:16 INFO     Running RISC-V instruction generator
Fri, 28 Aug 2020 18:07:16 INFO     Generating 2 riscv_rand_instr_test
Fri, 28 Aug 2020 18:07:16 INFO     Running riscv_rand_instr_test with 1 batches
Fri, 28 Aug 2020 18:07:16 INFO     Running riscv_rand_instr_test, batch 1/1, test_cnt:2
Fri, 28 Aug 2020 18:07:22 INFO     Compiling out_2020-08-28/asm_tests/riscv_rand_instr_test_0.S
Fri, 28 Aug 2020 18:07:22 INFO     Converting to out_2020-08-28/asm_tests/riscv_rand_instr_test_0.bin
Fri, 28 Aug 2020 18:07:22 INFO     Compiling out_2020-08-28/asm_tests/riscv_rand_instr_test_1.S
Fri, 28 Aug 2020 18:07:22 INFO     Converting to out_2020-08-28/asm_tests/riscv_rand_instr_test_1.bin
Fri, 28 Aug 2020 18:07:22 INFO     Processing ISS setup file : /home/danghai/hhoangda/riscv-dv/yaml/iss.yaml
Fri, 28 Aug 2020 18:07:22 INFO     Found matching ISS: spike
Fri, 28 Aug 2020 18:07:22 INFO     spike sim log dir: out_2020-08-28/spike_sim
Fri, 28 Aug 2020 18:07:22 INFO     Running spike sim: out_2020-08-28/asm_tests/riscv_rand_instr_test_0.o
Fri, 28 Aug 2020 18:07:22 INFO     Running spike sim: out_2020-08-28/asm_tests/riscv_rand_instr_test_1.o
```

Qrun:
```
danghai@ubuntu:~/hhoangda/riscv-dv$ python3 run.py --test riscv_rand_instr_test --simulator qrun
Fri, 28 Aug 2020 18:08:25 INFO     Creating output directory: out_2020-08-28
Fri, 28 Aug 2020 18:08:25 INFO     Processing regression test list : /home/danghai/hhoangda/riscv-dv/target/rv32imc/testlist.yaml, test: riscv_rand_instr_test
Fri, 28 Aug 2020 18:08:25 INFO     Processing regression test list : /home/danghai/hhoangda/riscv-dv/yaml/base_testlist.yaml, test: riscv_rand_instr_test
Fri, 28 Aug 2020 18:08:25 INFO     Found matched tests: riscv_rand_instr_test, iterations:2
Fri, 28 Aug 2020 18:08:25 INFO     Processing simulator setup file : /home/danghai/hhoangda/riscv-dv/yaml/simulator.yaml
Fri, 28 Aug 2020 18:08:25 INFO     Found matching simulator: qrun
Fri, 28 Aug 2020 18:08:25 INFO     Building RISC-V instruction generator
Fri, 28 Aug 2020 18:08:25 INFO     Running RISC-V instruction generator
Fri, 28 Aug 2020 18:08:25 INFO     Generating 2 riscv_rand_instr_test
Fri, 28 Aug 2020 18:08:25 INFO     Running riscv_rand_instr_test with 1 batches
Fri, 28 Aug 2020 18:08:25 INFO     Running riscv_rand_instr_test, batch 1/1, test_cnt:2
Fri, 28 Aug 2020 18:08:31 INFO     Compiling out_2020-08-28/asm_tests/riscv_rand_instr_test_0.S
Fri, 28 Aug 2020 18:08:31 INFO     Converting to out_2020-08-28/asm_tests/riscv_rand_instr_test_0.bin
Fri, 28 Aug 2020 18:08:31 INFO     Compiling out_2020-08-28/asm_tests/riscv_rand_instr_test_1.S
Fri, 28 Aug 2020 18:08:31 INFO     Converting to out_2020-08-28/asm_tests/riscv_rand_instr_test_1.bin
Fri, 28 Aug 2020 18:08:31 INFO     Processing ISS setup file : /home/danghai/hhoangda/riscv-dv/yaml/iss.yaml
Fri, 28 Aug 2020 18:08:31 INFO     Found matching ISS: spike
Fri, 28 Aug 2020 18:08:31 INFO     spike sim log dir: out_2020-08-28/spike_sim
Fri, 28 Aug 2020 18:08:31 INFO     Running spike sim: out_2020-08-28/asm_tests/riscv_rand_instr_test_0.o
Fri, 28 Aug 2020 18:08:31 INFO     Running spike sim: out_2020-08-28/asm_tests/riscv_rand_instr_test_1.o

```

udinator, 2020-08-31:
@aliess what version of Questa are you using?
In addition, there should be no need to modify the macro definitions to get the generator to run successfully.
Can you provide links to the simulation logs and the generated assembly programs?

aliess, 2020-09-02:
I am using questa 2018.08 revision.

you can see that there's another issue here that addresses the problem of the macro definitions for the same version (https://github.com/google/riscv-dv/issues/83)

is there a way I can debug the problem on my version ?

aliess, 2020-09-02:
I am attaching the files for the 2 runs, the riscv_arithmetic_basic_test run that succeeds and riscv_rand_instr_test that fails

[riscv_arithmetic_basic_test.tar.gz](https://github.com/google/riscv-dv/files/5161019/riscv_arithmetic_basic_test.tar.gz)
[riscv_rand_instr_test.tar.gz](https://github.com/google/riscv-dv/files/5161020/riscv_rand_instr_test.tar.gz)

aliess, 2020-09-18:
did anyone check this ? thanks

taoliug, 2020-09-20:
I pushed a fix for it. please let me know if you still see this issue.

closed 2020-09-20 by taoliug

aliess, 2020-09-21:
thank you for the reply.

your commit removed some errors from `riscv_jump_stress_test` but there is one fatal error
`UVM_FATAL /home/ali/Documents/riscv-dv/src/riscv_directed_instr_lib.sv(299) @ 0: reporter [instr_stack_enter] stack len [-1829382354] is not enough to store           1 regs`
[Archive.zip](https://github.com/google/riscv-dv/files/5254614/Archive.zip)

on the other hand `riscv_rand_instr_test` is still generating errors in the same file you changed.

will attach both logs.

aliess, 2020-10-13:
hey, any chance you can get another look at this ? 
thank you.

commit 9ec65d0f (2020-09-20): Fix #681 (#715)