chipsalliance/riscv-dv #686: there is no coverage data,when split to small batches and run with LSF in parallel

yinhanquan, 2020-08-24:
when we try to use the cd: cov --dir out/ovpsim_sim --lsf_cmd bsub -bz 5 --target rv64g,there is no test.vdb file ,but we use the cd: cov --dir out/ovpsim_sim  --target rv64g,the coverage result is ok .how can we solve this one if we have a large number of csv to collect coverage? thank you.

taoliug, 2020-08-30:
can you share the screen log when you run this command?

yinhanquan, 2020-08-31:
ok,as follows:
1.
[@ser140 riscv-dv]$ cov --dir cov_test --lsf_cmd "bsub -Is" -bz 5 --target rv64g -o ./cov_result
Sun, 30 Aug 2020 21:25:44 INFO     Creating output directory: ./cov_result
Sun, 30 Aug 2020 21:25:44 INFO     Processing trace log under cov_test
Sun, 30 Aug 2020 21:25:44 INFO     Process ovpsim log[1/35] : cov_test/rv64i_m_add_10.log
Sun, 30 Aug 2020 21:25:44 INFO     Processing ovpsim log : cov_test/rv64i_m_add_10.log
Sun, 30 Aug 2020 21:25:45 INFO     Processed instruction count : 10855 
Sun, 30 Aug 2020 21:25:45 INFO     CSV saved to : cov_test/rv64i_m_add_10.csv
Sun, 30 Aug 2020 21:25:45 INFO     Process ovpsim log[2/35] : cov_test/rv64i_m_add_11.log
Sun, 30 Aug 2020 21:25:45 INFO     Processing ovpsim log : cov_test/rv64i_m_add_11.log
Sun, 30 Aug 2020 21:25:45 INFO     Processed instruction count : 11494 
Sun, 30 Aug 2020 21:25:45 INFO     CSV saved to : cov_test/rv64i_m_add_11.csv
Sun, 30 Aug 2020 21:25:45 INFO     Process ovpsim log[3/35] : cov_test/rv64i_m_add_12.log
Sun, 30 Aug 2020 21:25:45 INFO     Processing ovpsim log : cov_test/rv64i_m_add_12.log
Sun, 30 Aug 2020 21:25:46 INFO     Processed instruction count : 10800 
Sun, 30 Aug 2020 21:25:46 INFO     CSV saved to : cov_test/rv64i_m_add_12.csv
Sun, 30 Aug 2020 21:25:46 INFO     Process ovpsim log[4/35] : cov_test/rv64i_m_add_13.log
Sun, 30 Aug 2020 21:25:46 INFO     Processing ovpsim log : cov_test/rv64i_m_add_13.log
Sun, 30 Aug 2020 21:25:46 INFO     Processed instruction count : 11490 
Sun, 30 Aug 2020 21:25:46 INFO     CSV saved to : cov_test/rv64i_m_add_13.csv
Sun, 30 Aug 2020 21:25:46 INFO     Process ovpsim log[5/35] : cov_test/rv64i_m_add_14.log
Sun, 30 Aug 2020 21:25:46 INFO     Processing ovpsim log : cov_test/rv64i_m_add_14.log
Sun, 30 Aug 2020 21:25:47 INFO     Processed instruction count : 10870 
Sun, 30 Aug 2020 21:25:47 INFO     CSV saved to : cov_test/rv64i_m_add_14.csv
Sun, 30 Aug 2020 21:25:47 INFO     Process ovpsim log[6/35] : cov_test/rv64i_m_add_15.log
Sun, 30 Aug 2020 21:25:47 INFO     Processing ovpsim log : cov_test/rv64i_m_add_15.log
Sun, 30 Aug 2020 21:25:47 INFO     Processed instruction count : 11340 
Sun, 30 Aug 2020 21:25:47 INFO     CSV saved to : cov_test/rv64i_m_add_15.csv
Sun, 30 Aug 2020 21:25:47 INFO     Process ovpsim log[7/35] : cov_test/rv64i_m_add_16.log
Sun, 30 Aug 2020 21:25:47 INFO     Processing ovpsim log : cov_test/rv64i_m_add_16.log
Sun, 30 Aug 2020 21:25:48 INFO     Processed instruction count : 11011 
Sun, 30 Aug 2020 21:25:48 INFO     CSV saved to : cov_test/rv64i_m_add_16.csv
Sun, 30 Aug 2020 21:25:48 INFO     Process ovpsim log[8/35] : cov_test/rv64i_m_add_17.log
Sun, 30 Aug 2020 21:25:48 INFO     Processing ovpsim log : cov_test/rv64i_m_add_17.log
Sun, 30 Aug 2020 21:25:48 INFO     Processed instruction count : 11241 
Sun, 30 Aug 2020 21:25:48 INFO     CSV saved to : cov_test/rv64i_m_add_17.csv
Sun, 30 Aug 2020 21:25:48 INFO     Process ovpsim log[9/35] : cov_test/rv64i_m_add_18.log
Sun, 30 Aug 2020 21:25:48 INFO     Processing ovpsim log : cov_test/rv64i_m_add_18.log
Sun, 30 Aug 2020 21:25:49 INFO     Processed instruction count : 11088 
Sun, 30 Aug 2020 21:25:49 INFO     CSV saved to : cov_test/rv64i_m_add_18.csv
Sun, 30 Aug 2020 21:25:49 INFO     Process ovpsim log[10/35] : cov_test/rv64i_m_add_19.log
Sun, 30 Aug 2020 21:25:49 INFO     Processing ovpsim log : cov_test/rv64i_m_add_19.log
Sun, 30 Aug 2020 21:25:49 INFO     Processed instruction count : 11076 
Sun, 30 Aug 2020 21:25:49 INFO     CSV saved to : cov_test/rv64i_m_add_19.csv
Sun, 30 Aug 2020 21:25:49 INFO     Process ovpsim log[11/35] : cov_test/rv64i_m_add_1.log
Sun, 30 Aug 2020 21:25:49 INFO     Processing ovpsim log : cov_test/rv64i_m_add_1.log
Sun, 30 Aug 2020 21:25:49 INFO     Processed instruction count : 10782 
Sun, 30 Aug 2020 21:25:49 INFO     CSV saved to : cov_test/rv64i_m_add_1.csv
Sun, 30 Aug 2020 21:25:49 INFO     Process ovpsim log[12/35] : cov_test/rv64i_m_add_20.log
Sun, 30 Aug 2020 21:25:49 INFO     Processing ovpsim log : cov_test/rv64i_m_add_20.log
Sun, 30 Aug 2020 21:25:50 INFO     Processed instruction count : 11283 
Sun, 30 Aug 2020 21:25:50 INFO     CSV saved to : cov_test/rv64i_m_add_20.csv
Sun, 30 Aug 2020 21:25:50 INFO     Process ovpsim log[13/35] : cov_test/rv64i_m_add_21.log
Sun, 30 Aug 2020 21:25:50 INFO     Processing ovpsim log : cov_test/rv64i_m_add_21.log
Sun, 30 Aug 2020 21:25:50 INFO     Processed instruction count : 10927 
Sun, 30 Aug 2020 21:25:50 INFO     CSV saved to : cov_test/rv64i_m_add_21.csv
Sun, 30 Aug 2020 21:25:50 INFO     Process ovpsim log[14/35] : cov_test/rv64i_m_add_22.log
Sun, 30 Aug 2020 21:25:50 INFO     Processing ovpsim log : cov_test/rv64i_m_add_22.log
Sun, 30 Aug 2020 21:25:51 INFO     Processed instruction count : 11443 
Sun, 30 Aug 2020 21:25:51 INFO     CSV saved to : cov_test/rv64i_m_add_22.csv
Sun, 30 Aug 2020 21:25:51 INFO     Process ovpsim log[15/35] : cov_test/rv64i_m_add_23.log
Sun, 30 Aug 2020 21:25:51 INFO     Processing ovpsim log : cov_test/rv64i_m_add_23.log
Sun, 30 Aug 2020 21:25:51 INFO     Processed instruction count : 10777 
Sun, 30 Aug 2020 21:25:51 INFO     CSV saved to : cov_test/rv64i_m_add_23.csv
Sun, 30 Aug 2020 21:25:51 INFO     Process ovpsim log[16/35] : cov_test/rv64i_m_add_24.log
Sun, 30 Aug 2020 21:25:51 INFO     Processing ovpsim log : cov_test/rv64i_m_add_24.log
Sun, 30 Aug 2020 21:25:52 INFO     Processed instruction count : 11523 
Sun, 30 Aug 2020 21:25:52 INFO     CSV saved to : cov_test/rv64i_m_add_24.csv
[... cut ...]
Sun, 30 Aug 2020 21:33:11 INFO     Coverage results are saved to ./cov_result
[@ser140 riscv-dv]$ cd cov_result/
[@ser140 cov_result]$ ls
asm_tests         sim_riscv_instr_cov_test_0_0.log  sim_riscv_instr_cov_test_0_4.log  vc_hdrs.h
compile.log       sim_riscv_instr_cov_test_0_1.log  sim_riscv_instr_cov_test_0_5.log  vcs_simv
ovpsim_trace_log  sim_riscv_instr_cov_test_0_2.log  sim_riscv_instr_cov_test_0_6.log  vcs_simv.csrc
seed.yaml         sim_riscv_instr_cov_test_0_3.log  test.vdb                          vcs_simv.daidir

closed 2020-11-02 by yinhanquan