chipsalliance/riscv-dv #687: Free RTL simulation tools for students

NazerkeT, 2020-08-25:
Hi there,
Recently, I have build support for a CVA6 core for 4 bitmanip subgroups and wanted to test them for validity. Coming to this point, I have found out that there is not actually any student licences that let full-stack RTL simulation. However, still want to believe to the good news and want to clarify :) Can I run this _riscv-dv_ using some student licence from third-party RTL simulators? Or is there any other similar opportunity for System Verilog verification?

Best regards,

udinator, 2020-08-26:
Hi @NazerkeT, if I understand you correctly, yes - to run the SystemVerilog riscv-dv generator you will need a license for any of the simulators that we support (VCS, IUS, Questa, DSim, Xcelium, or Riviera).

closed 2020-08-30 by taoliug

omerguzelelectronicguy, 2023-01-26:
Is there any option at least just to create a torture assembly test without any simulation?

NazerkeT, 2023-01-27:
@omerguzelelectronicguy So far nothing that I found or am aware of :(

But maybe in the last 2 years, there have been some advances in it, you might wanna search a bit more I guess 😅 

Best wishes,