chipsalliance/riscv-dv #690: jump to an unexpected return address, unexpected instruction address_misaligned exception occurs

syvenwang7, 2020-08-28:
gen_opts:
+instr_cnt=10000
+boot_mode=s
+enable_page_table_exception=1
+disable_compressed_instr=1
+directed_instr_0=riscv_load_store_rand_instr_stream,4
... (several directed load/store stream)

in a generated test, firstly, a load page fault exception occurs. And the exception handler firstly fix pte and decide to run some kernel program before exiting from exception handling.   j_smode -> smode_program. In smode_program_stack_p，store the gp register to stack. After smode program finishes, jump to gp. However, the gp is never updated after init (0xaf654e2a). jump instruction triggers instruction address misaligned exception.

taoliug, 2020-08-31:
Can you upload the log and generated assembly test for debugging? Thanks
For page table, are you using SV32 or SV39?

vandanaprabhu, 2020-09-01:
The `+enable_page_table_exception` knob is not stable as I have seen a similar issue during randomization of `riscv_page_table_exception_test` (20% see this issue) due to program execution not being able to return correctly from `smode_program`. As you can see from the code snippets below, `ra` gets a write value `0x00000000_8001356C` after  `jal  ra,80012238 <smode_program>`  but gets overwritten shortly after entering `smode_program` and consequently,  is unable to return on the ` c.jr    a4` instruction at the end of `smode_program` at `80012b1e`
My attempts to fix this locally by saving `ra` upon entry to `smode_program` resolved this issue but broke the generator code in other ways.  I use SV39 and I have attached the generator compile log and complete generated assembly file.

```
....
....
0000000080012238 <smode_program>:
    80012238:   41b65a3b                sraw    s4,a2,s11
    8001223c:   fd0a8a93                addi    s5,s5,-48
    80012240:   01f100b3                add     ra,sp,t6
    80012244:   9e9d                    c.subw  a3,a5
    80012246:   01bab423                sd      s11,8(s5)
    8001224a:   8e21                    c.xor   a2,s0
    8001224c:   e9080b13                addi    s6,a6,-368
    80012250:   02147d3b                remuw   s10,s0,ra
    80012254:   41be00b3                sub     ra,t3,s11
0000000080012258 <smode_program_1>:
    80012258:   00400def                jal     s11,8001225c <smode_program_1+0x4>
    8001225c:   004000ef                jal     ra,80012260 <smode_program_1+0x8>
    80012260:   01c000ef                jal     ra,8001227c <smode_program_1+0x24>
    80012264:   024000ef                jal     ra,80012288 <smode_program_1+0x30>
    80012268:   a021                    c.j     80012270 <smode_program_1+0x18>
....
    80012b04:   028779b3                remu    s3,a4,s0
    80012b08:   4f45                    c.li    t5,17
    80012b0a:   03696233                rem     tp,s2,s6
    80012b0e:   030a8a93                addi    s5,s5,48
    80012b12:   02ab53b3                divu    t2,s6,a0
    80012b16:   00b4d43b                srlw    s0,s1,a1
    80012b1a:   000d8713                addi    a4,s11,0 # ffffffff99bcb000 <_end+0xffffffff19b87300>
    80012b1e:   8702                    c.jr    a4                -----> fails to return correctly due to incorrect value in `ra`

0000000080012b20 <smode_lsu_program>:
    80012b20:   8e21                    c.xor   a2,s0
    80012b22:   f88a8a93                addi    s5,s5,-120
....
....

0000000080013552 <check_mprv>:
    80013552:   00020d37                lui     s10,0x20
    80013556:   300028f3                csrrs   a7,mstatus,zero
    8001355a:   01a8f8b3                and     a7,a7,s10
    8001355e:   00088563                beq     a7,zero,80013568 <j_smode>
    80013562:   dbeff0ef                jal     ra,80012b20 <smode_lsu_program>
    80013566:   a019                    c.j     8001356c <fix_pte_ret>

0000000080013568 <j_smode>:
    80013568:   cd1fe0ef                jal     ra,80012238 <smode_program>.  

---> Above executes correctly upto this point, saves  8001356c to ra correctly and jumps to the correct target address but fails to return to execute fix_pte_ret below  and the diag goes off into the weeds thereafter

000000008001356c <fix_pte_ret>:
    8001356c:   008ab083                ld      ra,8(s5)
    80013570:   010ab103                ld      sp,16(s5)
    80013574:   018ab183                ld      gp,24(s5)
    80013578:   020ab203                ld      tp,32(s5)
    8001357c:   028ab283                ld      t0,40(s5)
    80013580:   030ab303                ld      t1,48(s5)
    80013584:   038ab383                ld      t2,56(s5)
    80013588:   040ab403                ld      s0,64(s5)
....
...
....

[sim_riscv_page_table_exception_test_0.log](https://github.com/google/riscv-dv/files/5158044/sim_riscv_page_table_exception_test_0.log)

[riscv_page_table_exception_test.S.TXT](https://github.com/google/riscv-dv/files/5158058/riscv_page_table_exception_test.S.TXT)

```