chipsalliance/riscv-dv #709: Compliation error for test case riscv_rand_instr_test with instr_cnt of 200000

suhasskrishanmurthy, 2020-09-14:
Hi, 
  I changed instr_cnt to 200000 in file ./yaml/base_testlist.yaml. Now ASM files are being genegerated but compilation is throwing error(linker error).
Please kindly tell me, what is that I am doing wrong.
The report generated by riscv-dv is below for riscv_rand_instr_test test case.

Mon, 14 Sep 2020 10:19:24 INFO     Creating output directory: out_2020-09-14
Mon, 14 Sep 2020 10:19:24 INFO     Processing regression test list : ./target/rv64im/testlist.yaml, test: riscv_rand_instr_test
Mon, 14 Sep 2020 10:19:24 INFO     Processing regression test list : /home/ariane/tandem/test_vectors/riscv-dv/target/rv32imc/testlist.yaml, test: riscv_rand_instr_test
Mon, 14 Sep 2020 10:19:24 INFO     Processing regression test list : /home/ariane/tandem/test_vectors/riscv-dv/yaml/base_testlist.yaml, test: riscv_rand_instr_test
Mon, 14 Sep 2020 10:19:24 INFO     Found matched tests: riscv_rand_instr_test, iterations:2
Mon, 14 Sep 2020 10:19:24 INFO     Processing simulator setup file : /home/ariane/tandem/test_vectors/riscv-dv/yaml/simulator.yaml
Mon, 14 Sep 2020 10:19:24 INFO     Found matching simulator: questa
Mon, 14 Sep 2020 10:19:24 INFO     Building RISC-V instruction generator
Mon, 14 Sep 2020 10:19:40 INFO     Running RISC-V instruction generator
Mon, 14 Sep 2020 10:19:40 INFO     Generating 2 riscv_rand_instr_test
Mon, 14 Sep 2020 10:19:40 INFO     Running riscv_rand_instr_test with 1 batches
Mon, 14 Sep 2020 10:19:40 INFO     Running riscv_rand_instr_test, batch 1/1, test_cnt:2
Mon, 14 Sep 2020 10:27:26 INFO     Compiling out_2020-09-14/asm_tests/riscv_rand_instr_test_0.S

/tmp/ccKK42Yk.o: In function `mepc_setup':
(.text+0x4c): relocation truncated to fit: R_RISCV_JAL against `init_machine_mode'
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "run.py", line 1038, in <module>
    main()
  File "run.py", line 1020, in main
    args.gcc_opts, args.debug)
  File "run.py", line 381, in gcc_compile
    run_cmd_output(cmd.split(), debug_cmd = debug_cmd)
  File "/home/ariane/tandem/test_vectors/riscv-dv/scripts/lib.py", line 190, in run_cmd_output
    raise exc
  File "/home/ariane/tandem/test_vectors/riscv-dv/scripts/lib.py", line 187, in run_cmd_output
    output = subprocess.check_output(cmd)
  File "/usr/lib/python3.6/subprocess.py", line 356, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/home/ariane/tandem/toolchain/bin/riscv64-unknown-elf-gcc', '-static', '-mcmodel=medany', '-fvisibility=hidden', '-nostdlib', '-nostartfiles', 'out_2020-09-14/asm_tests/riscv_rand_instr_test_0.S', '-I/home/ariane/tandem/test_vectors/riscv-dv/user_extension', '-T/home/ariane/tandem/test_vectors/riscv-dv/scripts/link.ld', '-o', 'out_2020-09-14/asm_tests/riscv_rand_instr_test_0.riscv', '-march=rv64im', '-mabi=lp64']' returned non-zero exit status 1.

Thanks in advance

udinator, 2020-09-15:
Hi @suhasskrishanmurthy thanks for opening this issue. This issue is due to some incorrect jump behavior in the generator, I will be preparing a fix.

udinator, 2020-09-21:
Hi @suhasskrishanmurthy can you confirm whether this error is gone or is still appearing? Thanks.

suhasskrishanmurthy, 2020-09-23:
Hi @udinator  yes the issue is resolved. Thanks for the fix.
Can You tell ,Maximum value of instr_cnt, that we can assign? right now till 300000 instr is working fine.

taoliug, 2020-09-23:
I think 300000 is already a fairly large number. I'd prefer to have smaller instruction count per test and run the test with more iterations. This could make the test fast to generate and simulate.

closed 2020-09-23 by taoliug