chipsalliance/riscv-dv #713: Set the NUM_HARTS = 2

suzhaow, 2020-09-17:
Hi,
I want to generate dual-core riscv_dv program, set NUM_HARTS=2. 
However, when using the command ： python3 run.py --custom_target xxx/target/rv64gc --isa rv64imafdc --mabi lp64d
The following error occurred：
xxx asm_tests/riscv_arithmetic_basic_test_0.S:12532: Error : symbol `test_done` is already define ....
How can this be corrected ？  
Thanks.

udinator, 2020-09-17:
Hi @suzhaow, is the branch you are using currently synced to the top of tree?
Could you also provide some links to the simulation log and the generated assembly file as well?

suzhaow, 2020-09-18:
> Hi @suzhaow, is the branch you are using currently synced to the top of tree?
> Could you also provide some links to the simulation log and the generated assembly file as well?
> I am looking into this error from my side as well.

suzhaow, 2020-09-18:
Hi,
Q1: Yes. The project obtained from GitHub in May of this year. 
And a small amount of modification has been made according to my own processor.
Q2: Here are the error report file and assembly file I extracted.
[error.log](https://github.com/google/riscv-dv/files/5245634/error.log)
[riscv_arithmetic_basic_test_0.log](https://github.com/google/riscv-dv/files/5245638/riscv_arithmetic_basic_test_0.log)
Hope these will be useful！
Thanks.

taoliug, 2020-09-20:
Just tested multi-harts program at top of tree, and it works for me
python3 run.py -tn riscv_load_store_shared_mem_test --target multi_harts

Can you share the generated program?

taoliug, 2020-09-20:
never mind, I am checking riscv_arithmetic_basic_test_0.log you uploaded.

taoliug, 2020-09-20:
I have submitted a fix for it. However, i don't think we have fully tested muli-harts + PMP enable, please let me know if we see any other issue.

taoliug, 2020-09-23:
Marked it solved, please reopen if you still have any issue

closed 2020-09-23 by taoliug

suzhaow, 2020-09-23:
First of all, I am sorry, I am busy with other things these days, thank you for your reply, I will try your suggestion.

suzhaow, 2020-09-23:
python3 run.py -tn riscv_load_store_shared_mem_test --target multi_harts. This is useful advice.It also works for me.
However, I encountered a new problem when I modified riscv_core_setting. sv in multi_HARts:
riscv_instr_group_t supported_isa[$] = {RV32I, RV32M, RV32C, RV32A
replace
riscv_instr_group_t supported_isa[$] = {RV32I, RV32M, RV64I, RV64M, RV32C, RV64C, RV32A, RV64A, RV32F, RV64F, RV32D, RV64D}
Here are the error report file and assembly file I extracted.
[error.log](https://github.com/google/riscv-dv/files/5268139/error.log)
[riscv_core_setting.log](https://github.com/google/riscv-dv/files/5268140/riscv_core_setting.log)
[riscv_load_store_shared_mem_test_0.log](https://github.com/google/riscv-dv/files/5268141/riscv_load_store_shared_mem_test_0.log)
[riscv_load_store_shared_mem_test_1.log](https://github.com/google/riscv-dv/files/5268143/riscv_load_store_shared_mem_test_1.log)
Hope these will be useful！
Thanks.

haroonshafique-lm, 2020-09-24:
Hi @suzhaow 
I went through the error log you shared. 

`-march` and `-mabi` being used in the tests compilation do not match with the `riscv_instr_group_t supported_isa[$]` in `riscv_core_setting.sv`. The ISA you are using is 64bit and thus you need to use `-march=rv64gc -mabi=lp64` in your python3 run command. 

Please let me know if you face the same issue. Thanks

suzhaow, 2020-09-24:
Yes. Thank you for your reply. I forgot to mention that I also uesd the command : 
python3 run.py -tn riscv_load_store_shared_mem_test -- target_multi_harts -march = rv64gc --mabi=lp64d.
But it still makes the same mistakes.
[error_rv64.log](https://github.com/google/riscv-dv/files/5274721/error_rv64.log)

haroonshafique-lm, 2020-09-24:
`run.py` takes `march` information through `--isa` option. Try the following command and let me know again if it works for you or not. 

`python3 run.py -tn riscv_load_store_shared_mem_test --target multi_harts --isa=rv64gc --mabi=lp64d`

suzhaow, 2020-09-24:
Unfortunately, it still makes the same mistakes.

haroonshafique-lm, 2020-09-24:
```
haroonshafique@computerserver:~/Desktop/riscv_dv_clones/riscv_dv_200921$ python3 run.py -tn riscv_load_store_shared_mem_test --target multi_harts --isa=rv64gc --mabi=lp64d
Thu, 24 Sep 2020 14:55:12 INFO     Creating output directory: out_2020-09-24
Thu, 24 Sep 2020 14:55:12 INFO     Processing regression test list : /home/users/haroonshafique/Desktop/riscv_dv_clones/riscv_dv_200921/target/multi_harts/testlist.yaml, test: riscv_load_store_shared_mem_test
Thu, 24 Sep 2020 14:55:12 INFO     Processing regression test list : /home/users/haroonshafique/Desktop/riscv_dv_clones/riscv_dv_200921/yaml/base_testlist.yaml, test: riscv_load_store_shared_mem_test
Thu, 24 Sep 2020 14:55:12 INFO     Found matched tests: riscv_load_store_shared_mem_test, iterations:2
Thu, 24 Sep 2020 14:55:12 INFO     Processing simulator setup file : /home/users/haroonshafique/Desktop/riscv_dv_clones/riscv_dv_200921/yaml/simulator.yaml
Thu, 24 Sep 2020 14:55:12 INFO     Found matching simulator: vcs
Thu, 24 Sep 2020 14:55:12 INFO     Building RISC-V instruction generator
Thu, 24 Sep 2020 14:55:13 INFO     Running RISC-V instruction generator
Thu, 24 Sep 2020 14:55:13 INFO     Generating 2 riscv_load_store_shared_mem_test
Thu, 24 Sep 2020 14:55:13 INFO     Running riscv_load_store_shared_mem_test with 1 batches
Thu, 24 Sep 2020 14:55:13 INFO     Running riscv_load_store_shared_mem_test, batch 1/1, test_cnt:2
Thu, 24 Sep 2020 14:55:33 INFO     Compiling out_2020-09-24/asm_tests/riscv_load_store_shared_mem_test_0.S
Thu, 24 Sep 2020 14:55:34 INFO     Converting to out_2020-09-24/asm_tests/riscv_load_store_shared_mem_test_0.bin
Thu, 24 Sep 2020 14:55:34 INFO     Compiling out_2020-09-24/asm_tests/riscv_load_store_shared_mem_test_1.S
Thu, 24 Sep 2020 14:55:34 INFO     Converting to out_2020-09-24/asm_tests/riscv_load_store_shared_mem_test_1.bin
Thu, 24 Sep 2020 14:55:34 INFO     Processing ISS setup file : /home/users/haroonshafique/Desktop/riscv_dv_clones/riscv_dv_200921/yaml/iss.yaml
Thu, 24 Sep 2020 14:55:34 INFO     Found matching ISS: spike
Thu, 24 Sep 2020 14:55:34 INFO     spike sim log dir: out_2020-09-24/spike_sim
Thu, 24 Sep 2020 14:55:34 INFO     Running spike sim: out_2020-09-24/asm_tests/riscv_load_store_shared_mem_test_0.o
Thu, 24 Sep 2020 14:55:34 INFO     Running spike sim: out_2020-09-24/asm_tests/riscv_load_store_shared_mem_test_1.o
```

This is working fine for me.

haroonshafique-lm, 2020-09-24:
@suzhaow
Are you using the latest master?
Did you make changes other than `riscv_instr_group_t supported_isa[$]` in `riscv_core_setting.sv`?

suzhaow, 2020-09-24:
Q1 : It is not the latest.
Q2 : New changes : 
parameter int XLEN = 64;
privileged_mode_t supported_privileged_mode[] = {USER_MODE,SUPERVISOR_MODE,MACHINE_MODE};
riscv_instr_group_t supported_isa[$] = {RV32I, RV32M, RV64I, RV64M, RV32C, RV64C, RV32A, RV64A, RV32F, RV64F, RV32D, RV64D};
Do I need to update the project? Or there may be other solutions ？

haroonshafique-lm, 2020-09-24:
@suzhaow

In your case, actual problem is that you are modifying the **pre-defined** `multi_harts` target. This will not affect the default `--isa` and `--mabi` options even provided by you. If you want to have a modified version of a pre-defined target, then make a copy of `mutli_harts/` folder in the `target/` directory with other name let say `custom_multi_harts/`. This directory should contain `riscv_core_setting`, `testlist.yaml` and `riscvOVPsim.ic`. Then change the xlen, privilege modes and supported ISA in riscv_core_setting.sv.

After doing the above things properly, run the command with one more option `--core_setting_dir=target/custom_multi_harts` along `--isa=rv64gc --mabi=lp64d` as follows:

```
python3 run.py -tn riscv_load_store_shared_mem_test --target multi_harts --isa=rv64gc --mabi=lp64d --core_setting_dir=target/custom_multi_hart
```

suzhaow, 2020-09-24:
Nice!!! It solved my problem perfectly.Thanks for your help.

suzhaow, 2020-09-28:
Excuse me again.
I modified the src/riscv_asm_program_gen.sv based on your file.(Fix multi-harts program generation with PMP enabled #716)
But when setting riscv_core_setting.sv/bit support_pmp = 1, it has the following error:
/home/suzw/toolchain/install_multilib/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/cc4kMbgD. o: in function .L0 :
/home/suzw/toolchain/install_multilib/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: (.text+0x120d2 ): undefined reference to main
collect2: error: ld returned 1 exit status.
Am I missing something ?
Thanks.